RTimothyEdwards/magic #6: Convert the git tags to annotated

mithro, 2020-01-24:
So git-describe finds them.

mithro, 2020-02-04:
@RTimothyEdwards Can you run [the following script](https://gist.github.com/mithro/55fe2f4f872cb3c67f661425ab011004) on this repository? It should convert all the current tags to be annotated tags (and thus things like `git-describe` should work correctly with them).

RTimothyEdwards, 2020-02-04:
So this script will not fix anything going forward.  I have an automated python script that at the end of each day updates the git repository on opencircuitdesign.com, and if anything has been committed in the previous 24 hours, it automatically increments the revision number and runs "git tag <version>".  Apparently that does not make tags annotated.  So what do I need to do to automatically annotate the tags at the time of creating the tag?  Otherwise, this script will just put me into a vicious cycle where I check in the change to tags, and that check-in gets tagged with an unannotated tag, and 'round we go.

RTimothyEdwards, 2020-02-04:
Apparently all I need to do in my automated script is:
    git tag -a -m"Tagging version $version" $version
instead of
    git tag $version
?

mithro, 2020-02-04:
Yeap!

RTimothyEdwards, 2020-02-04:
What does it mean exactly to be "annotated"?

mithro, 2020-02-04:
> An annotated tag creates an additional tag object in the Git repository, which allows you to store information associated with the tag itself. This may include release notes, the meta-information about the release, and optionally a signature to verify the authenticity of the commit to which it points.

mithro, 2020-02-04:
From https://git-scm.com/book/en/v2/Git-Basics-Tagging
> ## Creating Tags
> Git supports two types of tags: **lightweight** and **annotated**.
> 
> A **lightweight** tag is very much like a branch that doesn’t change — it’s just a pointer to a specific commit.
> 
> **Annotated** tags, however, are stored as full objects in the Git database. They’re checksummed; contain the tagger name, email, and date; have a tagging message; and can be signed and verified with GNU Privacy Guard (GPG). It’s generally recommended that you create annotated tags so you can have all this information; but if you want a temporary tag or for some reason don’t want to keep the other information, lightweight tags are available too.

RTimothyEdwards, 2020-02-04:
Running the script in my local cloned copy does not create a change that I can commit.  Is it only meaningful to run this on the master repository?

mithro, 2020-02-04:
You might need to do a `git fetch origin --tags` first?

RTimothyEdwards, 2020-02-04:
No, the problem was only that I had not noticed that the script already does a push at the end, so there was nothing left to do after the script ran.  Although I did find that running the script twice deletes all the tags.  Fortunately that causes an error and apparently it does not push that change, because I re-cloned the repo and the tags were still there.  Fortunately.

RTimothyEdwards, 2020-02-04:
Done.  Will close this tomorrow after the mirror updates overnight.

closed 2020-02-14 by RTimothyEdwards