RTimothyEdwards/magic #8: [OpenBSD/ld.bfd] magicexec and magicdnull want GR_LIBS to link successfully

julianaito, 2020-02-17:
As i said in https://github.com/RTimothyEdwards/netgen/issues/2, building magic with the GNU ld.bfd found in old architecture also requires `GR_LIBS` for magicexec (and magicdnull), otherwise [it fails](http://build-failures.rhaalovely.net/powerpc/last/cad/magic.log).

Here are the changes needed. I added line breaks as lines were longer than 80 characters, i'm no sure if you want that, but i had to take a decision :)

(cc @just22 @kmosiejczuk)

```diff
Index: tcltk/Makefile
--- tcltk/Makefile.orig
+++ tcltk/Makefile
@@ -49,11 +49,13 @@ install-tcl: magicexec magicdnull ${BIN_FILES} ${TCL_F
 
 magicexec: magicexec.c ${MAGICDIR}/defs.mak
        ${CC} ${CFLAGS} ${CPPFLAGS} ${DFLAGS_NOSTUB} ${LDFLAGS} magicexec.c \
-               -o magicexec ${LD_RUN_PATH} ${LIBS} ${LIB_SPECS_NOSTUB}
+               -o magicexec \
+               ${LD_RUN_PATH} ${LIB_SPECS_NOSTUB} ${LIBS} ${GR_LIBS}
 
 magicdnull: magicdnull.c ${MAGICDIR}/defs.mak
        ${CC} ${CFLAGS} ${CPPFLAGS} ${DFLAGS_NOSTUB} ${LDFLAGS} magicdnull.c \
-               -o magicdnull ${LD_RUN_PATH} ${LIBS} ${LIB_SPECS_NOSTUB}
+               -o magicdnull \
+               ${LD_RUN_PATH} ${LIB_SPECS_NOSTUB} ${LIBS} ${GR_LIBS} 
 
 magic.tcl: magic.tcl.in ${MAGICDIR}/defs.mak
        sed -e /TCL_DIR/s%TCL_DIR%${TCLDIR}%g  \
```

RTimothyEdwards, 2020-02-17:
Same as for netgen:  Thanks, and patch has been applied, and update will mirror to github overnight.  I will close tomorrow after checking the update.

julianaito, 2020-02-17:
You're welcome!

RTimothyEdwards, 2020-02-18:
Somehow the mirror is not happening for magic (although it's working for netgen).  I need to investigate what happened. . .

RTimothyEdwards, 2020-02-18:
A previous pull request broke my automatic nightly mirror, so I just mirrored it manually.  Looks okay, so I am closing this issue.

closed 2020-02-18 by RTimothyEdwards