RTimothyEdwards/magic #9: proto.magicrc: Errors in make.log

FriendFX, 2020-02-19:
I just tried to compile on a Xubuntu 18.04 box and found the following errors in make.log:
```
proto.magicrc.in:1:3: error: invalid preprocessing directive #$
 # $(CAD_ROOT)/magic/sys/.magicrc
   ^
proto.magicrc.in:2:4: error: invalid preprocessing directive #System
 #  System wide start up file for magic, defines default macros.
    ^~~~~~
proto.magicrc.in:5:3: error: invalid preprocessing directive #Source
 # Source file proto.magicrc.in
   ^~~~~~
proto.magicrc.in:6:3: error: invalid preprocessing directive #Process
 # Process this file with the cpp macro processor
   ^~~~~~~
proto.magicrc.in:13:4: error: invalid preprocessing directive #Default
 #  Default .magicrc macro file (new macros)
    ^~~~~~~
proto.magicrc.in:15:3: error: invalid preprocessing directive #A
 # A key
   ^
proto.magicrc.in:19:3: error: invalid preprocessing directive #B
 # B key
   ^
proto.magicrc.in:22:3: error: invalid preprocessing directive #C
 # C key
   ^
proto.magicrc.in:24:3: error: invalid preprocessing directive #D
 # D key
   ^
proto.magicrc.in:27:3: error: invalid preprocessing directive #E
 # E key
   ^
proto.magicrc.in:29:3: error: invalid preprocessing directive #F
 # F key
   ^
proto.magicrc.in:32:3: error: invalid preprocessing directive #G
 # G key
   ^
proto.magicrc.in:35:3: error: invalid preprocessing directive #I
 # I key
   ^
proto.magicrc.in:41:3: error: invalid preprocessing directive #L
 # L key
   ^
proto.magicrc.in:50:3: error: invalid preprocessing directive #M
 # M key
   ^
proto.magicrc.in:53:3: error: invalid preprocessing directive #N
 # N key
   ^
proto.magicrc.in:55:3: error: invalid preprocessing directive #O
 # O key
   ^
proto.magicrc.in:58:3: error: invalid preprocessing directive #P
 # P key
   ^
proto.magicrc.in:65:3: error: invalid preprocessing directive #Q
proto.magicrc.in:69:3: error: invalid preprocessing directive #R
 # R key
   ^
proto.magicrc.in:73:3: error: invalid preprocessing directive #S
 # S key
   ^
proto.magicrc.in:80:3: error: invalid preprocessing directive #U
 # U key
   ^
proto.magicrc.in:83:3: error: invalid preprocessing directive #V
 # V key
   ^
proto.magicrc.in:86:3: error: invalid preprocessing directive #W
 # W key
   ^
proto.magicrc.in:89:3: error: invalid preprocessing directive #X
 # X key
   ^
proto.magicrc.in:93:3: error: invalid preprocessing directive #Z
 # Z key
   ^
proto.magicrc.in:100:3: error: invalid preprocessing directive #Question
 # Question mark
   ^~~~~~~~
proto.magicrc.in:103:3: error: invalid preprocessing directive #Comma
 # Comma key
   ^~~~~
proto.magicrc.in:105:3: error: invalid preprocessing directive #Exclamation
 # Exclamation mark
   ^~~~~~~~~~~
proto.magicrc.in:109:3: error: invalid preprocessing directive #Space
 # Space bar
   ^~~~~
proto.magicrc.in:111:3: error: invalid preprocessing directive #Colon
 # Colon and semicolon (interactive command)
   ^~~~~
proto.magicrc.in:117:3: error: invalid preprocessing directive #Arrow
 # Arrow keys (X11 versions only)
   ^~~~~
proto.magicrc.in:134:3: error: invalid preprocessing directive #Keypad
 # Keypad keys (X11 versions only)
   ^~~~~~
proto.magicrc.in:135:3: error: invalid preprocessing directive #Functions
 # Functions duplicated for use both with Num_Lock ON and OFF
   ^~~~~~~~~
proto.magicrc.in:167:3: error: invalid preprocessing directive #Scroll
 # Scroll wheel bindings
   ^~~~~~
proto.magicrc.in:170:3: error: invalid preprocessing directive #Quick
 # Quick macro function keys for scmos tech (X11 versions only)
   ^~~~~
proto.magicrc.in:233:3: error: invalid preprocessing directive #Allow
 # Allow some box manipulation from all tools.
   ^~~~~
proto.magicrc.in:238:3: error: invalid preprocessing directive #Box
 # Box tool button bindings
   ^~~
proto.magicrc.in:257:3: error: invalid preprocessing directive #Color
 # Color window button bindings
   ^~~~~
proto.magicrc.in:265:3: error: invalid preprocessing directive #Netlist
 # Netlist window button bindings
   ^~~~~~~
proto.magicrc.in:269:3: error: invalid preprocessing directive #Wind3D
 # Wind3D window key bindings
   ^~~~~~
proto.magicrc.in:298:3: error: invalid preprocessing directive #Load
 # Load basic set of fonts
   ^~~~
proto.magicrc.in:304:3: error: invalid preprocessing directive #Additions
 # Additions for Tcl GUI wrapper
   ^~~~~~~~~
Makefile:42: recipe for target 'proto.magicrc' failed
make[2]: *** [proto.magicrc] Error 1
```

RTimothyEdwards, 2020-02-19:
What in the world is Xubuntu doing with gcc??  Well, I have a workaround from another tool I've been developing, which is a Python replacement for the gcc preprocessor, which is not really suited to be used as a general-purpose macro processor.  As the gcc preprocessor expects the file it's working on to be C language, passing it a Tcl script does not work well.  I'll add in my python script and see how that goes.

RTimothyEdwards, 2020-02-19:
Well, I gave up on the C preprocessor long ago as hopeless for doing non-C-code related tasks, and added a Python preprocessor, but apparently I never applied that script to the macro definition file.  I have done so now and it appears to work.  I pushed the fix to the git repo on opencircuitdesign.com, and the github version should update overnight.  I will check it tomorrow;  but since I always have trouble with the preprocessing stuff, I'll let you confirm that it actually works right before I close this issue.

FriendFX, 2020-02-20:
Awesome thanks, that fixed it. 

I am slightly concerned about your comment "What in the world is Xubuntu doing with gcc??" but currently don't know enough about the build environment to really worry until something (else?) breaks.

closed 2020-02-20 by FriendFX