RTimothyEdwards/magic #10: Version inconsistencies b/w opencircuitdesign.com download page and github mirror

just22, 2020-02-23:
Hello Tim,

magic's OpenBSD port is picking the source tarball from opencircuitdesign.com download page. As of today (Sun, 23 Feb) the latest available version is 8.2.182 (timestamp: Jan 30, 2020 at 3:01am); please note that this version doesn't contain [this fix](https://github.com/RTimothyEdwards/magic/issues/8).

On github mirror I see [this](https://github.com/RTimothyEdwards/magic/commit/6049b7e00e288d7bb57d962300610281b21756da) as most recent commit (timestamp: Feb 17), the code correctly contains the above cited fix, but VERSION file reports 8.2.101.

Is this expected?

All the best

--
Alessandro

RTimothyEdwards, 2020-02-23:
It's not *expected*, but I'm aware of the problem.  I've been trying to fix it for several days now.  The whole automatic update thing stopped working suddenly at the end of January, and I found some problems with the cloned repo on opencircuitdesign.com that is used for generating the tarball.  I have been "fixing it" every day and letting the automatic process run overnight, and finding in the morning that it's still the same.  I will start running it manually until I figure out what's really wrong with it.

VERSION in the master branch only updates if I updated it manually in the master repo.  If the instructions on the site are followed and magic is compiled and installed from the magic-8.2 branch, then the VERSION is correct.  I hesitated to try an automatic update of VERSION in the master branch, but I guess I'll take a shot at implementing that.

---Tim

RTimothyEdwards, 2020-02-24:
I finally got it all sorted out;  it looks okay now and all repositories are consistent.  The VERSION file is now the same in all branches.  I'll close this for now.  If anything looks weirdly out of sync in the future, feel free to re-open it.

Thanks,
Tim

closed 2020-02-24 by RTimothyEdwards