RTimothyEdwards/magic #16: Generated `ANTENNAPARTIALMETALSIDEAREA` in the lef output

mithro, 2020-05-24:
@RTimothyEdwards recently added `ANTENNADIFFAREA` and `ANTENNAGATEAREA` to Magic's LEF output. However he didn't seem to add `ANTENNAPARTIALMETALSIDEAREA`.

RTimothyEdwards, 2020-05-24:
@mithro :  If you can read the LEF/DEF spec and tell me what the hell is meant by "ANTENNAPARTIALMETALSIDEAREA" (or "ANTENNAPARTIALMETALAREA"), I would be glad to compute it and add it to the output.  As it is, however, I read it four or five times and could not make any sense of it.  The metal area of the pin is the "geometric area" plus this value.  So this value seems to be some mystical thing that is not included in the geometric area of the pin.

RTimothyEdwards, 2020-05-24:
@mithro :  So trying to divine what is meant by ANTENNAPARTIALMETALSIDEAREA, I looked at the example cell you cited, and the value matches the perimeter sidewall area of metal1.  But the metal1 geometry of the pin is all right there in the PORT record, so that makes no sense with respect to the description in the LEF/DEF spec, because adding the physical geometry to the value of ATENNAPARTIALMETALSIDEAREA would then double-count the metal1.  So I have no understanding of when you would want to output this value, and why.

RTimothyEdwards, 2020-05-24:
@mithro :  Also that does not explain why the value includes only metal1, and not li1 or mcon areas.

RTimothyEdwards, 2020-05-28:
@mithro :  One thing that comes to mind is that if the "-hide" option is used, then the entire metal area of the pin may not be represented by the pin geometry.  In that case, the spec does make some sense, as the value output would be the sum total area of metal not seen in the abstract view, and it would be appropriate to add that value to the represented geometry to arrive at the total.  I could certainly compute and output that value.  That said, I'm pretty sure the example you showed me is using it incorrectly.  But I have thought things like that before and ended up concluding that the problem was in the spec, not in the example.