RTimothyEdwards/magic #18: Magic gets stuck waiting on null terminator

mithro, 2020-06-03:
Finally had the magic sitting at 100% issue happen again. Magic seems to be stuck waiting for a null terminator;
```
tansell@pdk:~/skywater-pdk-scratch-new$ gdb -p 62681
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 62681
[New LWP 63466]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
efReadLine (line=line@entry=0x7ffd0c696860 "", size=-288, size@entry=1024, file=file@entry=0x56343261c360, argv=argv@entry=0x7ffd0c696460) at EFread.c:695
695         while (*get != '\0')
(gdb) bt
#0  efReadLine (line=line@entry=0x7ffd0c696860 "", size=-288, size@entry=1024, file=file@entry=0x56343261c360, argv=argv@entry=0x7ffd0c696460) at EFread.c:695
#1  0x00007fcfcc9c3f20 in efReadDef (def=def@entry=0x56343440a130, dosubckt=dosubckt@entry=1 '\001', resist=resist@entry=0 '\000', noscale=noscale@entry=0 '\000', toplevel=toplevel@entry=1 '\001') at EFread.c:240
#2  0x00007fcfcc9c2c90 in EFReadFile (name=name@entry=0x563433fafb40 "sky130_fd_pr_rf2__rf2_xcmvpp11p5x11p7_m3_lim5shield", dosubckt=dosubckt@entry=1 '\001', resist=0 '\000', noscale=noscale@entry=0) at EFread.c:145
#3  0x00007fcfcc96467a in CmdExtToSpice (w=<optimized out>, cmd=0x56343438d140) at ext2spice.c:704
#4  0x00007fcfcc91705a in WindExecute (w=w@entry=0x563433371890, rc=94782184192688, cmd=cmd@entry=0x56343438d140) at windMain.c:414
#5  0x00007fcfcc89f45f in DBWcommands (w=0x563433371890, cmd=0x56343438d140) at DBWprocs.c:598
#6  0x00007fcfcc913c05 in WindSendCommand (w=0x563433371890, w@entry=0x0, cmd=cmd@entry=0x56343438d140, quiet=quiet@entry=1 '\001') at windSend.c:263
#7  0x00007fcfcc90c1c1 in TxTclDispatch (clientData=clientData@entry=0x0, argc=argc@entry=1, argv=argv@entry=0x563432637090, quiet=quiet@entry=1 '\001') at txCommands.c:1180
#8  0x00007fcfcca0c037 in _tcl_dispatch (clientData=0x0, interp=0x5634326291e0, argc=1, argv=0x563432637090) at tclmagic.c:400
#9  0x00007fcfcea531eb in TclInvokeStringCommand () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#10 0x00007fcfcea54fb7 in TclNRRunCallbacks () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#11 0x00007fcfcea563af in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#12 0x00007fcfceb0f3f8 in Tcl_FSEvalFileEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#13 0x00007fcfceb0dd57 in Tcl_EvalFile () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#14 0x00007fcfcc959304 in mainInitFinal () at main.c:1164
#15 0x00007fcfcca0caa8 in _magic_startup (clientData=0x0, interp=0x5634326291e0, argc=1, argv=0x563432636c50) at tclmagic.c:725
#16 0x00007fcfcea531eb in TclInvokeStringCommand () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#17 0x00007fcfcea54fb7 in TclNRRunCallbacks () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#18 0x00007fcfcea563af in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#19 0x00007fcfceb0f3f8 in Tcl_FSEvalFileEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#20 0x00007fcfceb0dd57 in Tcl_EvalFile () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#21 0x00007fcfceb16f6e in Tcl_SourceRCFile () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#22 0x00007fcfceb174bd in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#23 0x0000563431b3822d in main (argc=7, argv=0x7ffd0c697cc8) at magicdnull.c:49
(gdb) 
```

mithro, 2020-06-03:
```
(gdb) p *get
$1 = 0 '\000'
(gdb) p get
$2 = 0x7ffd0c696860 ""
(gdb) 
```

mithro, 2020-06-03:
```
(gdb) step
759         if (argc == 0)
(gdb) step
760             goto start;
(gdb) 
675          get = line;
(gdb) p line
$3 = 0x7ffd0c696860 ""
(gdb) step
676          while (size > 0)
(gdb) p size
$4 = -288
(gdb) step
689         if (size == 0) efReadError("long line truncated\n");
```

mithro, 2020-06-03:
```
(gdb) generate-core-file magic-stuck.core
warning: target file /proc/62681/cmdline contained unexpected null characters
Saved corefile magic-stuck.core
```

mithro, 2020-06-03:
```
/usr/local/lib/magic/tcl/magicdnull^@-nowrapper^@-dnull^@-noconsole^@-T^@/home/tansell/skywater-src-nda/s8.tech^@/home/tansell/tempdir/gds/s8/V1.0.0/gds/s8rf2/s8rf2.gds/gen_output.sky130_fd_pr_rf2__rf2_xcmvpp11p5x11p7_m3_lim5shield.tcl^@
```

mithro, 2020-06-03:
```
commit 336a7aa2098b6c076f07ae61f5db6f10cad4fc1c (HEAD -> master, upstream/master, origin/master)
Author: Tim Edwards <tim@opencircuitdesign.com>
Date:   Fri May 29 17:13:38 2020 -0400

    Added missing close-bracket in the readspice script, and corrected
    a diagnostic error output in the port command.
```

mithro, 2020-06-03:
```
(gdb) p line
$8 = 0x7ffd0c696860 ""
(gdb) p *line
$9 = 0 '\000'
```

mithro, 2020-06-03:
This seems to be the loop that it is stuck in;
```
760             goto start;
(gdb) 
675          get = line;
(gdb) 
676          while (size > 0)
(gdb) 
689         if (size == 0) efReadError("long line truncated\n");
(gdb) 
691         get = put = line;
(gdb) 
693         if (*line == '#') goto start;       /* Ignore comments */
(gdb) 
695         while (*get != '\0')
(gdb) 
759         if (argc == 0)
(gdb) p argc
$11 = 0
(gdb) step
760             goto start;
(gdb) 
(gdb) step
675          get = line;
(gdb) 
676          while (size > 0)
(gdb) 
689         if (size == 0) efReadError("long line truncated\n");
(gdb) 
691         get = put = line;
(gdb) 
693         if (*line == '#') goto start;       /* Ignore comments */
(gdb) 
695         while (*get != '\0')
(gdb) 
759         if (argc == 0)
(gdb) 
760             goto start;
(gdb) 
675          get = line;
(gdb) 
676          while (size > 0)
(gdb) 
689         if (size == 0) efReadError("long line truncated\n");
(gdb) 
691         get = put = line;
(gdb) 
693         if (*line == '#') goto start;       /* Ignore comments */
(gdb) 
695         while (*get != '\0')
(gdb) 
759         if (argc == 0)
(gdb) 
760             goto start;
(gdb) 
675          get = line;
(gdb) 
676          while (size > 0)
(gdb) 
689         if (size == 0) efReadError("long line truncated\n");
(gdb) 
691         get = put = line;
(gdb) 
693         if (*line == '#') goto start;       /* Ignore comments */
(gdb) 
695         while (*get != '\0')
(gdb) 
759         if (argc == 0)
(gdb) 
760             goto start;
(gdb) 
675          get = line;
(gdb) 
676          while (size > 0)
(gdb) 
689         if (size == 0) efReadError("long line truncated\n");
```

RTimothyEdwards, 2020-06-04:
@mithro :  Should be prevented by changing line 759 to "if ((argc == 0) && (size > 0))".  However, it appears that the underlying cause is a line overflow, which is probably caused by some node name being ridiculously long, perhaps?  The line overflow isn't going to do this routine any good, and probably it needs a dynamically allocated buffer for the read.  But at least take the code that I just pushed and see what comes of it.

closed 2020-06-04 by RTimothyEdwards