RTimothyEdwards/magic #28: Magic segfault while doing spice extraction

mithro, 2020-07-19:
`magic -nowrapper -dnull -noconsole -T /ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8.tech /ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl`

[s8Dp_blkinv_opt1a.gds.tar.gz](https://github.com/RTimothyEdwards/magic/files/4943872/s8Dp_blkinv_opt1a.gds.tar.gz)

```tcl
#!/bin/env wish
drc off
scalegrid 1 2
gds readonly true
gds rescale false
tech unlock *
cif warning default
set VDD VPWR
set GND VGND
gds read /ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/sky130_fd_sp_sram__sram_dp_blkinv_mcon.gds
load {sky130_fd_sp_sram__sram_dp_blkinv_mcon}
lef write {sky130_fd_sp_sram__sram_dp_blkinv_mcon} -toplayer
load {sky130_fd_sp_sram__sram_dp_blkinv_mcon}
extract do local
extract all
ext2spice lvs
ext2spice
quit -noprompt
```

```
ERROR: Magic exited with status -11

magic -nowrapper -dnull -noconsole -T /ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8.tech /ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl
===========================================================================

Magic 8.3 revision 36 - Compiled on Sun Jul 19 07:23:25 PDT 2020.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Input style vendorimport: scaleFactor=2, multiplier=2
Processing system .magicrc file
Loading "/ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl" from command line.
2 Magic internal units = 1 Lambda
Warning: Calma reading is not undoable!  I hope that's OK.
Library written using GDS-II Release 3.0
Library name: sky130_fd_sp_sram__sram_dp_blkinv_mcon
Reading "sky130_fd_sp_sram__sram_dp_mcon_05".
Reading "sky130_fd_sp_sram__sram_dp_mcon_1".
Reading "sky130_fd_sp_sram__sram_dp_blkinv_mcon".
Generating LEF output sky130_fd_sp_sram__sram_dp_blkinv_mcon.lef for cell sky130_fd_sp_sram__sram_dp_blkinv_mcon:
Diagnostic:  Write LEF header for cell sky130_fd_sp_sram__sram_dp_blkinv_mcon
Diagnostic:  Writing LEF output for cell sky130_fd_sp_sram__sram_dp_blkinv_mcon
Diagnostic:  Scale value is 0.005000
Extracting sky130_fd_sp_sram__sram_dp_mcon_05 into sky130_fd_sp_sram__sram_dp_mcon_05.ext:
Extracting sky130_fd_sp_sram__sram_dp_mcon_1 into sky130_fd_sp_sram__sram_dp_mcon_1.ext:

===========================================================================
/ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds
/ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl
---------------------------------------------------------------------------
ERROR: Magic exited with status -11
magic                                                     :
magic                                                     : Magic 8.3 revision 36 - Compiled on Sun Jul 19 07:23:25 PDT 2020.
magic                                                     : Starting magic under Tcl interpreter
magic                                                     : Using the terminal as the console.
magic                                                     : Using NULL graphics device.
magic                                                     : Input style vendorimport: scaleFactor=2, multiplier=2
magic                                                     : Processing system .magicrc file
magic                                                     : Loading "/ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb.tcl" from command line.
magic                                                     : 2 Magic internal units = 1 Lambda
magic                                                     : Warning: Calma reading is not undoable!  I hope that's OK.
magic                                                     : Library written using GDS-II Release 3.0
magic                                                     : Library name: sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: Reading "sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb".
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: Generating LEF output sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb.lef for cell sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb:
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: Diagnostic:  Write LEF header for cell sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: Diagnostic:  Writing LEF output for cell sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: Diagnostic:  Scale value is 0.005000
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: Extracting sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb into sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb.ext:
magic sky130_fd_sp_sram__sram_dp_swldrv_opt1_poly_siz_optb: exttospice finished.
```

mithro, 2020-07-19:
```
tim@desktop:~/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds$ gdb --args /opt/magic/lib/magic/tcl/magicdnull -nowrapper -noconsole -T /ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8.tech /ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl
GNU gdb (Ubuntu 8.2-0ubuntu1~18.04) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/magic/lib/magic/tcl/magicdnull...done.
(gdb) run
Starting program: /opt/magic/lib/magic/tcl/magicdnull -nowrapper -noconsole -T /ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8.tech /ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Magic 8.3 revision 36 - Compiled on Sun Jul 19 07:23:25 PDT 2020.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using TrueColor, VisualID 0x21 depth 24
Input style vendorimport: scaleFactor=2, multiplier=2
Processing system .magicrc file
Loading "/ssd/gob/foss-eda-tools/tempdir-openram/gds/ssd/gob/foss-eda-tools/skywater-src-nda-openram/s8_sram/V0.0.0/cells/s8Dp_blkinv_opt1a/s8Dp_blkinv_opt1a.gds/gen_output.sky130_fd_sp_sram__sram_dp_blkinv_mcon.tcl" from command line.
2 Magic internal units = 1 Lambda
Warning: Calma reading is not undoable!  I hope that's OK.
Library written using GDS-II Release 3.0
Library name: sky130_fd_sp_sram__sram_dp_blkinv_mcon
Reading "sky130_fd_sp_sram__sram_dp_mcon_05".
Reading "sky130_fd_sp_sram__sram_dp_mcon_1".
Reading "sky130_fd_sp_sram__sram_dp_blkinv_mcon".
Generating LEF output sky130_fd_sp_sram__sram_dp_blkinv_mcon.lef for cell sky130_fd_sp_sram__sram_dp_blkinv_mcon:
Diagnostic:  Write LEF header for cell sky130_fd_sp_sram__sram_dp_blkinv_mcon
Diagnostic:  Writing LEF output for cell sky130_fd_sp_sram__sram_dp_blkinv_mcon
Diagnostic:  Scale value is 0.005000
Extracting sky130_fd_sp_sram__sram_dp_mcon_05 into sky130_fd_sp_sram__sram_dp_mcon_05.ext:
Extracting sky130_fd_sp_sram__sram_dp_mcon_1 into sky130_fd_sp_sram__sram_dp_mcon_1.ext:

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff5245c41 in extHierConnectFunc1 (oneTile=oneTile@entry=0x7ffff7e717a0, ha=ha@entry=0x7fffffffcad0) at ExtHier.c:299
299             if (GEO_TOUCH(&r, &lab->lab_rect))
(gdb) bt
#0  0x00007ffff5245c41 in extHierConnectFunc1 (oneTile=oneTile@entry=0x7ffff7e717a0, ha=ha@entry=0x7fffffffcad0) at ExtHier.c:299
#1  0x00007ffff520ae24 in DBSrPaintArea (hintTile=hintTile@entry=0x0, plane=0x5555577cc1d0, rect=rect@entry=0x7fffffffcbc0, mask=0x7ffff5a1c1a0 <DBAllButSpaceBits>,
    func=func@entry=0x7ffff5245753 <extHierConnectFunc1>, arg=arg@entry=140737488341712) at DBtiles.c:439
#2  0x00007ffff52455b1 in extHierConnections (ha=ha@entry=0x7fffffffcad0, cumFlat=cumFlat@entry=0x7fffffffcae8, oneFlat=oneFlat@entry=0x5555574e2630) at ExtHier.c:191
#3  0x00007ffff524e071 in extSubtreeFunc (scx=0x7ffffff874e0, ha=0x7fffffffcad0) at ExtSubtree.c:804
#4  0x00007ffff51e3c5c in dbCellSrFunc (use=use@entry=0x55555678e4f0, cxp=cxp@entry=0x7fffffffc8f0) at DBcellsrch.c:1165
#5  0x00007ffff51e1d6e in DBSrCellPlaneArea (plane=0x5555573d4af0, rect=rect@entry=0x7fffffffc9d0, func=func@entry=0x7ffff51e374e <dbCellSrFunc>, arg=arg@entry=140737488341232) at DBcellsrch.c:91
#6  0x00007ffff51e3728 in DBCellSrArea (scx=scx@entry=0x7fffffffc9c0, func=func@entry=0x7ffff524dbd2 <extSubtreeFunc>, cdarg=cdarg@entry=140737488341712) at DBcellsrch.c:1117
#7  0x00007ffff524d51f in extSubtreeInteraction (ha=ha@entry=0x7fffffffcad0) at ExtSubtree.c:463
#8  0x00007ffff524d0f4 in extSubtree (parentUse=0x5555559473b0, reg=reg@entry=0x5555577c69c0, f=f@entry=0x555555a4d6f0) at ExtSubtree.c:250
#9  0x00007ffff5241c4e in extCellFile (def=def@entry=0x55555678e9f0, f=f@entry=0x555555a4d6f0, doLength=doLength@entry=0 '\000') at ExtCell.c:265
#10 0x00007ffff524176c in ExtCell (def=def@entry=0x55555678e9f0, outName=outName@entry=0x0, doLength=doLength@entry=0 '\000') at ExtCell.c:118
#11 0x00007ffff5249878 in extExtractStack (stack=0x5555577be9e0, doExtract=doExtract@entry=1, rootDef=0x55555678f1f0) at ExtMain.c:645
#12 0x00007ffff5248e26 in ExtAll (rootUse=rootUse@entry=0x5555577c66c0) at ExtMain.c:195
#13 0x00007ffff51c4978 in CmdExtract (w=<optimized out>, cmd=0x55555678b9f0) at CmdE.c:1032
#14 0x00007ffff529fadb in WindExecute (w=w@entry=0x5555558b6720, rc=93824996781360, cmd=cmd@entry=0x55555678b9f0) at windMain.c:414
[... cut ...]
#28 0x00007ffff7a5ffba in ?? () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#29 0x00007ffff7b19da8 in Tcl_FSEvalFileEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#30 0x00007ffff7b18637 in Tcl_EvalFile () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#31 0x00007ffff7b2173e in Tcl_SourceRCFile () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#32 0x00007ffff7b21d41 in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#33 0x0000555555554912 in main (argc=6, argv=0x7fffffffdc48) at magicdnull.c:49
(gdb) 
```

mithro, 2020-07-19:
It seems like `lab` is a null pointer;
```
(gdb) p r
$1 = {r_ll = {p_x = -18, p_y = -18}, r_ur = {p_x = 18, p_y = 1}}
(gdb) p lab
$2 = (Label *) 0x0
```

mithro, 2020-07-19:
```
(gdb) p cumDef
$4 = (CellDef *) 0x555555946eb0
(gdb) p *cumDef
$5 = {cd_flags = 73, cd_bbox = {r_ll = {p_x = 0, p_y = 0}, r_ur = {p_x = 2, p_y = 2}}, cd_extended = {r_ll = {p_x = 0, p_y = 0}, r_ur = {p_x = 2, p_y = 2}}, cd_file = 0x0, cd_fd = -1, 
  cd_name = 0x55555585ece0 "__ext_cumulative", cd_parents = 0x5555559473b0, cd_cellPlane = 0x5555559ab630, cd_planes = {0x55555585eda0, 0x55555582b150, 0x5555558b7ca0, 0x5555558c6b60, 0x55555582afd0, 
    0x5555558b7be0, 0x5555558b77a0, 0x55555594b500, 0x55555594b580, 0x555556797360, 0x555556798760, 0x555556770b10, 0x555556737ac0, 0x555556771c50, 0x55555679b8b0, 0x5555567976e0, 0x55555676ca40, 
    0x5555567739d0, 0x555555949c00, 0x0 <repeats 45 times>}, cd_client = 0, cd_timestamp = 0, cd_labels = 0x0, cd_lastLabel = 0x0, cd_technology = 0x0, cd_props = 0, cd_filler = 0, cd_idHash = {
    ht_table = 0x555555a0e890, ht_size = 16, ht_nEntries = 0, ht_downShift = 26, ht_mask = 15, ht_ptrKeys = 0, ht_copyFn = 0x0, ht_compareFn = 0x0, ht_hashFn = 0x0, ht_killFn = 0x0}, cd_types = {tt_words = {0, 
      0, 0, 0, 0, 0, 0, 0}}}
(gdb) 
```

mithro, 2020-07-19:
```
(gdb) p cumDef->cd_labels
$6 = (Label *) 0x0
```

mithro, 2020-07-19:
```patch
diff --git a/extract/ExtHier.c b/extract/ExtHier.c
index 7f7e58e..9413dea 100644
--- a/extract/ExtHier.c
+++ b/extract/ExtHier.c
@@ -289,6 +289,8 @@ extHierConnectFunc1(oneTile, ha)
     // Definitely this causes problems in arrays, because the array node
     // name may refer to a range of array elements, and the generated
     // node only describes a single point.
+    if (cumDef->cd_labels == NULL)
+        return (0);
 
     for (lab = cumDef->cd_labels;  lab;  lab = lab->lab_next)
 
```

mithro, 2020-07-19:
I'm a bit confused why the for loop isn't terminating due to `lab == NULL`

mithro, 2020-07-19:
It looks like a case of missing brackets?

```patch
diff --git a/extract/ExtHier.c b/extract/ExtHier.c
index 7f7e58e..2e571f5 100644
--- a/extract/ExtHier.c
+++ b/extract/ExtHier.c
@@ -290,7 +290,11 @@ extHierConnectFunc1(oneTile, ha)
     // name may refer to a range of array elements, and the generated
     // node only describes a single point.
 
-    for (lab = cumDef->cd_labels;  lab;  lab = lab->lab_next)
+    for (lab = cumDef->cd_labels;  lab;  lab = lab->lab_next) {
+
+       if (lab == NULL) {
+           break;
+       }
 
        // All sticky labels are at the front of the list in cumDef, so
        // stop when we see the first non-sticky label.
@@ -354,6 +358,7 @@ extHierConnectFunc1(oneTile, ha)
                }
 #endif
            }
+    }
 
     return (0);
 }
```

RTimothyEdwards, 2020-07-19:
Yep.  I expanded the body of the "for" loop from one line to two in the last commit and failed to add the brackets.  The check for "lab == NULL" in the middle is unnecessary, though.  I am pushing a fix now.  I was traveling part of the day and so I have just seen this now.