RTimothyEdwards/magic #29: Sometimes tech value in extracted spice netlist is (null)

mithro, 2020-07-20:
Normally the output looks like this;
```
* NGSPICE file created from sky130_fd_sc_hvl__lsbuflv2hv_clkiso_hlkg_3.ext - technology: sky130A
```

But sometimes I'm getting
```
* NGSPICE file created from sky130_fd_sc_hvl__lsbuflv2hv_clkiso_hlkg_3.ext - technology: (null)
```

I can't find a reproducible test case which causes it to occur however. I'm always using the same script and technology file.

It seems to happen more often while splitting the ram files....

mithro, 2020-07-20:
Example;
```patch
diff --git a/libraries/sky130_fd_pr_rf2/v0.10.1/cells/rf2_xcmvpp6p8x6p1_lim4shield/sky130_fd_pr_rf2__rf2_xcmvpp6p8x6p1_lim4shield.spice b/libraries/sky130_fd_pr_rf2/v0.10.1/cells/rf2_xcmvpp6p8x6p1_lim4shield/sky130_fd_pr_rf2__rf2_xcmvpp6p8x6p1_lim4shield.spice
index 58f9d2da2a6..aec536b9b4c 100644
--- a/libraries/sky130_fd_pr_rf2/v0.10.1/cells/rf2_xcmvpp6p8x6p1_lim4shield/sky130_fd_pr_rf2__rf2_xcmvpp6p8x6p1_lim4shield.spice
+++ b/libraries/sky130_fd_pr_rf2/v0.10.1/cells/rf2_xcmvpp6p8x6p1_lim4shield/sky130_fd_pr_rf2__rf2_xcmvpp6p8x6p1_lim4shield.spice
@@ -1,7 +1,5 @@
-* NGSPICE file created from sky130_fd_pr_rf2__rf2_xcmvpp6p8x6p1_lim4shield.ext - technology: (null)
+* NGSPICE file created from sky130_fd_pr_rf2__rf2_xcmvpp6p8x6p1_lim4shield.ext - technology: sky130A
```

mithro, 2020-07-20:
FYI - This is probably blocking the skywater low level primitive release.

RTimothyEdwards, 2020-07-20:
@mithro :  What is the difference in the environment above between the correct and incorrect examples?  In principle, magic always knows what the technology is and always writes that into the .ext file, which then copies it into the .spice file.  DBTechName (global variable) is always set;  the default is "minimum", not NULL, so I don't know how the value ends up as "(null)".  How can I reproduce this result?

mithro, 2020-07-20:
That is the problem, I can't reliably reproduce the effect. It seems to happen randomly from the same input....

RTimothyEdwards, 2020-07-26:
@mithro :  There is no way for the DBTechName to be NULL under normal operation, so this appears to indicate some kind of memory corruption, and presumably is related to other problems like devices appearing or disappearing from the netlist.  Probably it can be captured by running valgrind on magic during extraction.  "ext2spice" is not implicated as the "(null)" value appears in the .ext file after running "extract".

mithro, 2020-07-26:
Can you give me the valgrind command that I should run magic using?

RTimothyEdwards, 2020-07-27:
@mithro :  It should just be "valgrind magic -dnull <other arguments>".  If the error is obvious, valgrind will report it.  There is a more complicated use of valgrind + gdb that I use for difficult-to-find issues, with simple instructions that can be found here:

https://heeris.id.au/2016/valgrind-gdb/

In short, run "valgrind --vgdb-error=0 magic -dnull ..." which will print a statement about how to run gdb, and then block at main().  Follow the printed instructions in a separate terminal.  When it hits a memory access error, gdb will break on the error and you can get a backtrace.

mithro, 2020-07-27:
I just want a command which runs magic inside valgrind and then if memory corruption has been detected dumps a core and reports a failure exit code....

RTimothyEdwards, 2020-07-28:
Just running "valgrind magic <arguments>" will do everything except dump core.  You can parse the output for indications of where invalid accesses occurred.  Often that's sufficient to find the error.

mithro, 2020-07-29:
Running valgrind didn't detect anything it seems, no leaks were detected. I wonder if there is a race condition that is not happening with valgrind being enabled.

Going to run the valgrind version 5 times and look at the output to see if it continues to vary....