RTimothyEdwards/magic #39: Misleading DRC errors

praharshapm, 2020-08-29:
Hi,
I have a layout which on viewing in magic didn't show any DRC errors (on the top right side of the window), but when I select the whole design and type `drc check`, then I can see there are some DRC errors.
Due to this, initially I was under the impression that there are no DRC errors in the layout which was misleading.

RTimothyEdwards, 2020-08-31:
Thanks for the report.  I have been noticing this problem while working on DRC-testing layouts.  It seems to be limited to the SkyWater PDK tech file, so I believe that it has something to do with the halo (area of effect for DRC rules), but I'm not certain.  I will be looking into it.

RTimothyEdwards, 2020-09-16:
I believe that I have just fixed this error.  The actual error appears to have been introduced around 2008 and apparently only shows up in the SkyWater PDK due to the particular relationship between the DRC halo and the DRC step size.  At least, that's my current theory.  I have run a series of checks and it appears not to make DRC errors disappear any more.  The code has been updated on the opencircuitdesign.com repo master branch and will update in github overnight.  Existing layouts which had errors disappear and then were saved might continue to have the errors missing until a "drc check" is run on the entire design, due to the incremental nature of magic's DRC engine.  I will leave this issue open for a little bit until I am fairly certain that the error has in fact been corrected.