RTimothyEdwards/magic #40: Can't extract resistor

cardosorapha, 2020-09-10:
Hi, I've created a (really dumb) cell just to test out the resistance extraction capacities, consisting of an N channel FET and long poly lines. I followed all steps in [tut8](http://opencircuitdesign.com/magic/tutorials/tut8.html), which explains how to work with resistances and I managed to create a SPICE file with all parasitics, but it's just impossible with this circuit (followed the exact same steps).

I don't know if this is actually a bug or not, or if this is the best place to post it, but here's a picture of my circuit, and I'm using Magic 8.3.50.

Thanks.
[image]

cardosorapha, 2020-09-11:
I now realize that my labels should be ports, which I found out by reading archived emails and ctrl-F searching for "extresist". My problem doesn't exist anymore, thanks for keeping old emails in the website.

RTimothyEdwards, 2020-09-11:
I really need to update some of the tutorials with newer information.  The original extresist would only extract nets between transistor terminals, so a resistor ending in a label would be completely ignored.  I did not attempt to mess with that general principle of ignoring dangling wires going to nowhere, but I did add the capability to treat a port as a driver or receiver point.

I'm glad the email archive was useful (even if the documentation was lacking)!

cardosorapha, 2020-09-11:
Something still bothers me... the behavior is not really consistent as I thought so. I created another simple circuit with ports and "extresist" crashed everytime I ran it, but when I created a complete circuit, with VDD, GND, NMOS and PMOS, it worked fine with the ports.

[image]

Also, the tutorial file (tut8r.mag) doesn't use any ports whatsoever and it still extracts resistances as I expected, without segmentation faults.

One more thing I noticed: "ext2sim" running in "example.ext" sometimes creates "example.ext.sim" and sometimes "example.sim" (same thing happens with .nodes whenever labels are on).

RTimothyEdwards, 2020-09-11:
If your layout crashes extresist, please post the layout file, as I would like to use it to debug the (first) problem.  I am inclined not to worry about the ".ext.sim" problem because my goal (which I hope to get around to soon, because it is becoming a high priority) is to get rid of the dependence of "extresist" on the .sim files completely. The .ext files have all the needed information, and the .sim files are missing some needed information.  This goal would also simplify the full-parasitic extraction to just a few commands.

cardosorapha, 2020-09-11:
Here are the files of a layout that crashes extresist.
[crash.zip](https://github.com/RTimothyEdwards/magic/files/5209314/crash.zip)

RTimothyEdwards, 2020-09-11:
The "Gnd" connection is a limitation of the SCMOS technology files. . . they are old and do not have support for some things like substrate extraction that I added to Magic.  But it is also a limitation of the .sim format, which has no concept of a substrate at all (all .sim transistors are 3-terminal devices).  Which is why I want to remove the dependence on the .sim format.

closed 2020-11-17 by cardosorapha