RTimothyEdwards/magic #42: Extraction of mosfet devices with source and drain connected to same net.

d-m-bailey, 2020-10-16:
1) It appears that the default mosfet extraction can not extract devices with the source and drain connected to the same net.

While doing a device level extraction of the openlane spm design, the extract log can not extract decap (decoupling capacitor) cells.

Here is the output:

```
Extracting sky130_fd_sc_hd__decap_8 into sky130_fd_sc_hd__decap_8.ext:
Warning: No matching extraction type for device at (79 47)
Warning: No matching extraction type for device at (79 323)

```
There appears to be code in magic's extract/ExtBasic.c to handle incorrect number of terminals, but I don't think it's working.
Search for 'No matching extraction type for device' to find the general location.

2) It is possible to add the following lines to the sky130A tech file and extract the devices.

```
fet pfethvt pdiff,pdiffres,pdc 1 sky130_fd_pr__pfet_01v8_hvt error nwell 0 0
fet scnfet,nfet ndiff,ndiffres,ndc 1 sky130_fd_pr__nfet_01v8 error pwell,space/w 0 0 

```
However, the length and width (l and w) are extracted in reverse. Here's the extracted spice subcircuit.

```
.subckt sky130_fd_sc_hd__decap_6 VGND VNB VPB VPWR
$ ** SOURCE/DRAIN TIED
M1000 VPWR VGND VPWR VPB sky130_fd_pr__pfet_01v8_hvt w=1.97e+06u l=870000u
+ ad=0p pd=0u as=0p ps=0u
$ ** SOURCE/DRAIN TIED
M1001 VGND VPWR VGND VNB sky130_fd_pr__nfet_01v8 w=1.97e+06u l=550000u
+ ad=0p pd=0u as=0p ps=0u
.ends

```
The capacitor width should be smaller than the length (distance from source to drain).

RTimothyEdwards, 2020-10-17:
@d-m-bailey :  The "fet" type for extraction is a backwards-compatible thing for the SCMOS tech files from way back, and it *will* compute L and W incorrectly in many instances.  The "device mosfet" has much better extraction capabilities.  In this case, though, the SkyWater model is a subcircuit, so the device type "device msubcircuit" is needed.

That said, though, I can confirm your result with the decap cell, that it does not extract properly.  This should not happen, and it seems that my expansion of the extraction capabilities that I did recently has somehow broken the behavior;  it should not care that there is only one terminal found for FET devices.  I will investigate the problem and fix it.

RTimothyEdwards, 2020-10-17:
@d-m-bailey :  This has now been fixed.  Please confirm and I will close the ticket.

d-m-bailey, 2020-10-19:
Tim,

Thanks for the quick update.
Unfortunately, I don't have a setup to compile magic at the moment. And I
have to get a presentation ready by Saturday, but maybe next week I can
confirm the fix.

Mitch

d-m-bailey, 2020-11-05:
Still getting this error in openlane:rc4

```
Extracting sky130_fd_sc_hd__decap_8 into sky130_fd_sc_hd__decap_8.ext:
Warning:  No matching extraction type for device at (79 47)
Warning:  No matching extraction type for device at (79 323)
```
The magic version in the container is 8.3.64

RTimothyEdwards, 2020-11-05:
@d-m-bailey :  I can't vouch for what's in the container.  The fix was done in revision 8.3.66 or 67.

d-m-bailey, 2020-11-05:
@RTimothyEdwards : OK. I'll keep checking.

d-m-bailey, 2020-11-13:
@RTimothyEdwards As of 8.3.79, there are no errors with mosfet capacitor extraction.

However, the extract log contained the following warnings for decap cells.
```
Extracting sky130_fd_sc_hd__decap_8 into sky130_fd_sc_hd__decap_8.ext:
sky130_fd_sc_hd__decap_8: 2 warnings
```
I was unable to locate any details about the warnings.

RTimothyEdwards, 2020-11-13:
@d-m-bailey :  After extraction, warnings are left in the feedback entries.  You can list all the warnings with "feedback why" (select the top cell first), parse through the warnings with "feedback find", or dump the feedback contents with "feedback save <filename>".  I believe that the warning is only mentioning the fact that the source and drain are tied.

mooredan, 2020-11-13:
Here's part of a TCL script that I use to do extractions.  It'll loop over the warnings:

```
load ${CELL}
extract warn all
extract
feedback count
set fcnt [feedback count]
for {set i 1} {$i <= $fcnt} {incr i} {
   feedback find $i
   # feedback why
   puts [box values] 
}
```
I hope that you'll find it useful.

d-m-bailey, 2020-11-13:
@RTimothyEdwards @mooredan Thanks. That should be good enough, then.

As far as I'm concerned, this can be closed.

RTimothyEdwards, 2020-11-14:
Okay, will do.

closed 2020-11-14 by RTimothyEdwards