RTimothyEdwards/magic #44: Incorrect extraction of spice model for sram GDS file

mithro, 2020-11-13:
The GDS file at https://foss-eda-tools.googlesource.com/skywater-pdk-scratch/+/refs/heads/ram-audit-v0.0-1245-gd62012d_magic-8.3.80_2020-11-12_055155_0000/libraries/sky130_fd_bd_sram/v0.0.0/cells/openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.gds currently extracts as;
```spice
.subckt sky130_fd_bd_sram__openram_cell_6t VNB VPB VPB
X0 VPB a_146_104# a_146_182# VPB sky130_fd_pr__pfet_01v8_hvt w=140000u l=150000u
X1 a_146_182# WL a_146_182# VPB sky130_fd_pr__pfet_01v8_hvt w=70000u l=95000u
X2 BL WL a_146_104# SUBS sky130_fd_pr__special_nfet_latch w=140000u l=150000u
X3 GND a_146_104# a_146_182# SUBS sky130_fd_pr__special_nfet_latch w=210000u l=150000u
X4 a_146_104# a_146_182# GND SUBS sky130_fd_pr__special_nfet_latch w=210000u l=150000u
X5 a_146_104# WL a_146_104# VPB sky130_fd_pr__pfet_01v8_hvt w=70000u l=95000u
X6 a_146_182# WL BR SUBS sky130_fd_pr__special_nfet_latch w=140000u l=150000u
X7 a_146_104# a_146_182# VPB VPB sky130_fd_pr__pfet_01v8_hvt w=140000u l=150000u
.ends
```

The first issue is the pins on the top level are incorrect. They should be something like;
```
BR
BL
VDD
GND
WL
VPB
VNB
```
See https://foss-eda-tools.googlesource.com/skywater-pdk-scratch/+/refs/heads/ram-audit-v0.0-1245-gd62012d_magic-8.3.80_2020-11-12_055155_0000/libraries/sky130_fd_bd_sram/v0.0.0/cells/openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.gds.pins

RTimothyEdwards, 2020-11-14:
@mithro  :  I'll assume that you understand that the issue with the pins is coming from the OpenRAM GDS itself.

That leaves the matter of the core pFET device being misinterpreted as a non-core pFET-HVT, which is the right device but doesn't have the right model bins.  The problem is not in magic but in open_pdks, as the tech file for magic is not checking the HVTP mask for the device, and so not recognizing it.

The second problem is that the tech file was not attempting to distinguish between the smaller pass transistor nFET device and the larger latch transistor nFET device. 

I just pushed a commit to open_pdks to fix those two problems, here:

`commit 8d30fd3e7d8fc6d7bd64ed4f82a9b54baa899c7b`

mithro, 2020-11-14:
Is that a new `SKY130A.tech` file?

RTimothyEdwards, 2020-11-14:
Yes.  I just copied it to skywater-src-nda and pushed it.

mithro, 2020-11-14:
Using a `sky130A.tech` with md5sum `f165d5b4251881342f1e0d28b52b43f3` I'm still getting the same `.spice` output file.

mithro, 2020-11-14:
The .mag file did seem to change however,
```
diff --git a/libraries/sky130_fd_bd_sram/v0.0.0/cells/openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.mag b/libraries/sky130_fd_bd_sram/v0.0.0/cells/openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.mag
index 8065c03c1a0..f55d1fd3f28 100644
--- a/libraries/sky130_fd_bd_sram/v0.0.0/cells/openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.mag
+++ b/libraries/sky130_fd_bd_sram/v0.0.0/cells/openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.mag
@@ -371,7 +371,7 @@ flabel metal1 s 200 208 228 244 0 FreeSans 100 0 0 0 BL
 flabel metal2 s 592 217 623 251 0 FreeSans 2000 0 0 0 VDD
 flabel metal2 s -122 138 -91 172 0 FreeSans 2000 0 0 0 GND
 flabel metal2 s 234 59 265 93 0 FreeSans 2000 0 0 0 WL
-flabel nbase s 358 31 368 41 0 FreeSans 40 90 0 0 VPB
+flabel nwell s 358 31 368 41 0 FreeSans 40 90 0 0 VPB
 port 1 nsew
 flabel pwell s 132 34 142 46 0 FreeSans 100 90 0 0 VNB
 port 0 nsew
```

mithro, 2020-11-14:
```
#!/bin/env wish
drc off
scalegrid 1 2
cif istyle vendorimport
gds readonly true
gds rescale false
tech unlock *
cif warning default
set VDD VPWR
set GND VGND
set SUB SUBS
gds read /home/tansell/tempdir-openram/gds/home/tansell/skywater-src-nda-openram/s8sram/V0.0.0/cells/sram_cell_6t/cell_6t.gds/sky130_fd_bd_sram__openram_cell_6t/sky130_fd_bd_sram__openram_cell_6t.gds
select top cell
port renumber
load {sky130_fd_bd_sram__openram_cell_6t}
save
lef write {sky130_fd_bd_sram__openram_cell_6t} -toplayer
load {sky130_fd_bd_sram__openram_cell_6t}
extract do local
extract no all
extract all
ext2spice lvs
ext2spice
quit -noprompt
```

mithro, 2020-11-14:
[sky130_fd_bd_sram__openram_cell_6t.tar.gz](https://github.com/RTimothyEdwards/magic/files/5541398/sky130_fd_bd_sram__openram_cell_6t.tar.gz)

mithro, 2020-11-14:
Using `sky130A.tech` with md5sum of `b51965f9f218425ed3ab2c709d594e7c` from commit 2eed96af073a06f26bcdfcb6768bee2b48f824e2 to src-nda.

I now get;
```spice
.subckt sky130_fd_bd_sram__openram_cell_6t VNB VPB VPB
X0 a_146_182# WL BR SUBS sky130_fd_pr__special_nfet_pass w=140000u l=150000u
X1 GND a_146_104# a_146_182# SUBS sky130_fd_pr__special_nfet_latch w=210000u l=150000u
X2 BL WL a_146_104# SUBS sky130_fd_pr__special_nfet_pass w=140000u l=150000u
X3 a_146_104# a_146_182# GND SUBS sky130_fd_pr__special_nfet_latch w=210000u l=150000u
.ends
```

mithro, 2020-11-14:
If I remove the `cif istyle vendorimport` line from the tcl script I get;
```spice
.subckt sky130_fd_bd_sram__openram_cell_6t VNB VPB VPB
X0 a_146_182# WL BR SUBS sky130_fd_pr__special_nfet_pass w=140000u l=150000u
X1 a_146_182# WL a_146_182# VPB sky130_fd_pr__special_pfet_pass w=70000u l=95000u
X2 a_146_104# WL a_146_104# VPB sky130_fd_pr__special_pfet_pass w=70000u l=95000u
X3 GND a_146_104# a_146_182# SUBS sky130_fd_pr__special_nfet_latch w=210000u l=150000u
X4 BL WL a_146_104# SUBS sky130_fd_pr__special_nfet_pass w=140000u l=150000u
X5 a_146_104# a_146_182# GND SUBS sky130_fd_pr__special_nfet_latch w=210000u l=150000u
X6 VPB a_146_104# a_146_182# VPB sky130_fd_pr__special_pfet_pass w=140000u l=150000u
X7 a_146_104# a_146_182# VPB VPB sky130_fd_pr__special_pfet_pass w=140000u l=150000u
.ends
```

RTimothyEdwards, 2020-11-15:
@mithro :  There is one more commit I made to skywater-src-nda shortly after the above post that corrects the problem without you needing to remove the "cif istyle vendorimport" line.  It is still high up on my to-do list to figure out exactly how SkyWater is specifying ports and non-port labels in the GDS and adapt magic to it.  I've mentioned before that GDS does not actually have a way to describe ports;  there is a "standard hack" that I've seen used, but SkyWater appears to be doing it in a different way that I've never seen before.  Once I've sorted that out I should be able to get rid of the alternative input style, and just read and generate labels and ports like SkyWater does.

RTimothyEdwards, 2020-11-16:
@mithro :  Is this confirmed okay now?

mithro, 2020-11-16:
It seems so.

closed 2020-11-16 by mithro