RTimothyEdwards/magic #45: Capacitance extraction for node "a" depends upon the position of label "b"

mooredan, 2020-11-14:
Cells ppcap_tc1 and ppcap_tc2 are identical in
every way except for the position of label b
on metal1.  The cell is a poly2/poly capacitor,
with node "a" being the bottom plate (poly) and
node "b" being the top plate (poly2).

Both plates are contacted to metal1.

For convenience, a screenshot of each cell is
provided:

ppcap_tc1.png
ppcap_tc2.png

[image]

[image]

The difference in the .mag files:

```
$ diff ppcap_tc1.mag ppcap_tc2.mag
20c20
< rlabel metal1 s 19 10 19 10 2 b
---
> rlabel metal1 s 23 20 23 20 2 b

```

A TCL file "doit.tcl" is included to extract the two
cells. And to reproduce, delete the .ext files and
run the script:

```
$ rm -f ppcap{1,2}.ext
$ /usr/local/bin/magic -T SCN3ME_SUBM.30.tech -noconsole -dnull < doit.tcl > magic.log 2>&1
```

After extraction, here is the difference between the two
.ext files:

```
$ diff ppcap_tc{1,2}.ext
8c8
< port "b" 2 19 10 19 10 m1
---
> port "b" 2 23 20 23 20 m1
10c10
< node "b" 114 532.8 19 10 m1 0 0 0 0 0 0 0 0 0 0 0 0 176 60 0 0 64 40 0 0 0 0
---
> node "b" 114 532.8 23 20 m1 0 0 0 0 0 0 0 0 0 0 0 0 176 60 0 0 64 40 0 0 0 0
12,13c12,13
< cap "b" "a" 14878.1
< subcap "a" -3276.96
---
> cap "a" "b" 14878.1
> subcap "a" -1906.56
```

The significant (and electrical) difference is the subcap lines.  This is the
amount of capacitance added from the capacitance on node "a".  After translation
to a spice file, the capacitance between node "a" and the substrate
will be different.  The same design should produce the same electrical
parameters.  Label position should not affect this.

Tech file: SCN3ME_SUBM.30.tech is from
https://github.com/RTimothyEdwards/qflow/blob/master/tech/osu050/SCN3ME_SUBM.30.tech

Magic 8.3 revision 82

Complete, self-contained test case:
[magic_tc1.tar.gz](https://github.com/RTimothyEdwards/magic/files/5541569/magic_tc1.tar.gz)

Dan Moore
mooredan@suncup.net
Nov 14, 2020

mooredan, 2020-11-14:
BTW, I dug into this over the past several days.  I observed that the clipArea variable that is passed to the extFindCoupling function in extract/ExtCouple.c was different between the two extractions.   Tried following it back further but got lost.  Just how does one run gdb on magic?

RTimothyEdwards, 2020-11-16:
"magic" is a shell script, so you need to run gdb on the base executable, which is "magicexec" and is found in the install directory under lib/magic/tcl/magicexec.  Finding anything related to extFindCoupling is a royal pain, just be warned.  I am buried in preparation for tomorrow's FOSSi talk, but I may have time to look at it after that.  There are a couple more pressing issues with magic that are critical to the SkyWater shuttle run, so I may continue to be preempted by other emergencies for the remainder of November.

mooredan, 2020-12-06:
This doesn't appear to be dependent upon the technology file. I rewrote the scmos tech file to be more "modern" leveraging off of the skywater tech file (1 lamba = 10nm).  Looking into this issue again....
BTW, I did get gdb running.

mooredan, 2020-12-06:
Maybe a major clue:  If the labels are not ports (just plain labels) then the position of the label has no effect on the extraction capacitances.  How are the ports, versus labels, being treated differently in extraction?

mooredan, 2020-12-06:
Adding "extract no adjust" to the extraction routine actually results in the SPICE subckt that is desired. This is regardless of whether a label is a port or not.  The default behavior is to adjust.  

Using this option takes care of my immediate issue, but doesn't address the whole "moving label b changes a's capacitance" behavior.  Maybe this will expose itself again while going up the hierarchy?