RTimothyEdwards/magic #52: Add "minimum required Magic version" to tech file

mithro, 2020-12-05:
It would be good if tech files could specify the minimum version of Magic required for the tech file to be loaded correctly. Magic could then check that it's version is newer than the requested version in the tech file before attempting to read it (and thus help prevent things like it segfaulting on invalid stuff). With this functionality, Magic could then give a useful error message like "Please update your version of Magic to atleast XXYYZZ to use this technology file".

I think it would also be perfectly fine for the required magic line to be required to be the first executable line in a tech file (and if specified later in the tech file a hard error even if the magic version is okay).

RTimothyEdwards, 2020-12-16:
@mithro :  This is now implemented in magic-8.3.99, and the latest open_pdks makes use of it.  Going forward, it will be easier to handle versioning between magic and the techfile.

mithro, 2020-12-16:
Awesome!

RTimothyEdwards, 2021-01-04:
This is done and working, so I'm closing the issue now.

closed 2021-01-04 by RTimothyEdwards