RTimothyEdwards/magic #55: .gitconfig is part of the repository...

mooredan, 2020-12-08:
...how to handle local customizations that are repository-specific?

Not a magic issue, but a git use model question.   The file .gitconfig has been added to the repository, but local customization, which are repo-specific, are desired.  Since this file is part of the repository, any local changes to it show up in git diff.  And these local changes are not necessary applicable to any arbitrary clone of the repository.   Admittedly, I'm not a git expert and do not know how this situation should be handled.  

I'm not certain that I recall a local .gitconfig is normally part of the repository, maybe local customizations could be handled from ~/.gitconfig? ...I don't know....

e.g.

```
mooredan@macbookpro:~/src/magic$ git status
On branch sandbox
Your branch is up to date with 'origin/sandbox'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   .gitconfig

no changes added to commit (use "git add" and/or "git commit -a")
mooredan@macbookpro:~/src/magic$ git diff
diff --git a/.gitconfig b/.gitconfig
index d4f587d..3478db8 100644
--- a/.gitconfig
+++ b/.gitconfig
@@ -3,3 +3,11 @@
        clean = sed -e's/\\$Header:[^$]*\\$/\\$Header\\$/g'
 [diff "gds"]
         textconv = hexdump -v -C
+
+[branch "sandbox"]
+    remote = ocd
+    merge = refs/heads/master
+
+[branch "master"]
+    remote = ocd
+    merge = refs/heads/master
```

mithro, 2020-12-08:
`~/.gitconfig` or `<repo>/.git/config` should probably be used for local configurations. `<repo>/.gitconfig` is use for repository specific setup like the smudge filters.

RTimothyEdwards, 2020-12-08:
@mithro :  I can only assume that it got in there unnoticed with somebody's pull request, because I didn't put it there, and there is no such file in any of my other repositories.

RTimothyEdwards, 2020-12-08:
@mithro :  The file history says it came from May 23, which is when I applied _your_ patch to remove whitespace. . .

mooredan, 2020-12-08:
Thanks Tim, Looks like my local customization needs to go in <repo>/.git/config vs. <repo>/.gitconfig. 

This article, if I understand it correctly, https://stackoverflow.com/questions/18329621/how-to-store-a-git-config-as-part-of-the-repository, tells us that we may need to do this individually if we want to have the filter and the custom diff command for gds files.

`git config --local include.path ../.gitconfig`

RTimothyEdwards, 2020-12-08:
But, I would like to keep this issue open until I have actually committed the change removing the .gitignore directory.

mithro, 2020-12-08:
@RTimothyEdwards - You mean `.gitconfig` and you actually want to keep it for the `$Id$` support.

RTimothyEdwards, 2020-12-09:
@mithro :  Right, I meant .gitconfig.  But why do I need it, since I don't have it in any other repository that I'm aware of, with apparently no consequences?

mithro, 2020-12-09:
@RTimothyEdwards You need it for the `$Id$` expansion support.

mooredan, 2021-02-20:
I think that this issue can be closed.