RTimothyEdwards/magic #60: Segmentation fault under xrdp

mooredan, 2021-01-02:
Starting magic through a xrdp session (CentOS Stream 8 running xrdp.service) results in a Segmentation fault:

mooredan@centos:~$ magic -noconsole
Use openwrapper to create a new GUI-based layout window
Use closewrapper to remove a new GUI-based layout window

Magic 8.3 revision 108 - Compiled on Sat Jan  2 09:32:04 PST 2021.
Starting magic under Tcl interpreter
Using the terminal as the console.
None of TrueColor 15, 16 or 24, or PseudoColor 8 found. Cannot initialize DISPLAY :11.0
Segmentation fault (core dumped)

Normally, I run magic without the -noconsole option, but am using it to capture the error. Otherwise, the tkcon window just comes and goes.

I did verify that magic starts normally when logged in at the console -- it's just not working in a remote session.

This could be an issue with xrdp configuration on the CentOS machine.

mooredan, 2021-01-02:
A solution has been found:

- In the /etc/xrdp/xrdp.ini file, under the [Xvnc] section was a setting "xserverbpp=24" that was commented out. Uncommentting this line and restarting the xrdp service resolved the issue.

I added some debug code and found that the depth member in the XVisualInfo structure that is retreived by the XGetVisualInfo call in the GrTkInit functionin graphics/grTk1.c file had a value of 32.  

There's a comment in this file that depth of 32 is likely to work, but has not been tested.

The message before the core dump indicates that there's a problem, maybe magic should exit gracefully at that point.

mooredan, 2021-01-03:
pull request #61 takes care of the seg fault in this situation

RTimothyEdwards, 2021-01-04:
I merged pull request #61, so I'll go ahead and close this issue.

closed 2021-01-04 by RTimothyEdwards