RTimothyEdwards/magic #69: Implicit declaration of function is invalid in C99

jay20162016, 2021-02-20:
When building magic on mac, I get 
```
--- errors and warnings logged in file make.log
--- making header file database/database.h
--- making dependencies
--- making modules
--- compiling bplane/bpEnum.o
In file included from bpEnum.c:39:
In file included from bpEnum.c:39:
bpEnum.c:76:5: error: implicit declaration of function 'GeoCanonicalRect' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling cmwind/CMWcmmnds.o
In file included from CMWcmmnds.c:32:
In file included from CMWcmmnds.c:32:
--- compiling commands/CmdSubrs.o
In file included from CmdSubrs.c:33:
In file included from CmdSubrs.c:33:
CmdSubrs.c:409:14: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
CmdSubrs.c:409:14: note: place parentheses around the assignment to silence this warning
CmdSubrs.c:409:14: note: use '==' to turn this assignment into an equality comparison
CmdSubrs.c:433:9: error: implicit declaration of function 'DBTechNameTypes' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
CmdSubrs.c:433:9: note: did you mean 'DBTechNameType'?
CmdSubrs.c:755:8: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
CmdSubrs.c:755:8: note: place parentheses around the assignment to silence this warning
CmdSubrs.c:755:8: note: use '==' to turn this assignment into an equality comparison
--- compiling database/DBbound.o
In file included from DBbound.c:26:
In file included from DBbound.c:26:
DBbound.c:61:9: error: implicit declaration of function 'DBSrCellPlaneArea' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling dbwind/DBWcommands.o
In file included from DBWcommands.c:29:
In file included from DBWcommands.c:29:
--- compiling dbwind/DBWdisplay.o
In file included from DBWdisplay.c:34:
In file included from DBWdisplay.c:34:
DBWdisplay.c:652:5: error: implicit declaration of function 'GrBox' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DBWdisplay.c:815:2: error: implicit declaration of function 'GeoTransPointDelta' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DBWdisplay.c:815:2: note: did you mean 'GeoTransPoint'?
DBWdisplay.c:837:14: error: implicit declaration of function 'GeoTransAngle' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DBWdisplay.c:897:2: error: implicit declaration of function 'GrFontText' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DBWdisplay.c:897:2: note: did you mean 'GrPutText'?
DBWdisplay.c:1640:2: error: implicit declaration of function 'TxError' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling drc/DRCarray.o
In file included from DRCarray.c:32:
In file included from DRCarray.c:32:
--- compiling drc/DRCbasic.o
In file included from DRCbasic.c:33:
In file included from DRCbasic.c:33:
DRCbasic.c:345:2: error: implicit declaration of function 'DBResetTilePlane' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:349:5: error: implicit declaration of function 'drcCifCheck' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:423:7: error: implicit declaration of function 'drcCheckAngles' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:434:7: error: implicit declaration of function 'drcCheckAngles' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:539:4: error: implicit declaration of function 'drcCheckArea' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:610:8: error: implicit declaration of function 'drcCheckMaxwidth' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:616:8: error: implicit declaration of function 'drcCheckMaxwidth' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:628:4: error: implicit declaration of function 'drcCheckRectSize' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:638:7: error: implicit declaration of function 'drcCheckOffGrid' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
DRCbasic.c:1017:7: error: implicit declaration of function 'drcCheckOffGrid' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling extflat/EFargs.o
EFargs.c:186:3: error: implicit declaration of function 'TxError' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
EFargs.c:214:3: error: implicit declaration of function 'efSymAdd' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
EFargs.c:235:3: error: implicit declaration of function 'efSymAddFile' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
EFargs.c:235:3: note: did you mean 'efSymAdd'?
EFargs.c:214:3: note: 'efSymAdd' declared here
EFargs.c:284:3: error: implicit declaration of function 'TxPrintf' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
EFargs.c:294:7: error: implicit declaration of function 'TxError' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
EFargs.c:332:5: error: implicit declaration of function 'TxError' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling extract/ExtArray.o
In file included from ExtArray.c:33:
In file included from ExtArray.c:33:
ExtArray.c:170:12: error: implicit declaration of function 'DBCellSrArea' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
[... 377 lines cut ...]
gcrColl.c:394:3: error: implicit declaration of function 'gcrMoveTrack' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling tcltk/tclmagic.o
In file included from tclmagic.c:30:
In file included from tclmagic.c:30:
tclmagic.c:538:9: error: implicit declaration of function 'mainInitBeforeArgs' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:539:9: error: implicit declaration of function 'mainDoArgs' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:560:9: error: implicit declaration of function 'mainInitAfterArgs' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:560:9: note: did you mean 'mainInitBeforeArgs'?
tclmagic.c:538:9: note: 'mainInitBeforeArgs' declared here
tclmagic.c:725:9: error: implicit declaration of function 'mainInitFinal' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:932:2: error: implicit declaration of function 'txGetFileCommand' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling magic/magicTop.o
magicTop.c:53:5: error: implicit declaration of function 'magicMain' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
--- compiling tcltk/tclmagic.o
In file included from tclmagic.c:30:
In file included from tclmagic.c:30:
tclmagic.c:538:9: error: implicit declaration of function 'mainInitBeforeArgs' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:539:9: error: implicit declaration of function 'mainDoArgs' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:560:9: error: implicit declaration of function 'mainInitAfterArgs' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:560:9: note: did you mean 'mainInitBeforeArgs'?
tclmagic.c:538:9: note: 'mainInitBeforeArgs' declared here
tclmagic.c:725:9: error: implicit declaration of function 'mainInitFinal' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
tclmagic.c:932:2: error: implicit declaration of function 'txGetFileCommand' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
```
[make.log](https://github.com/RTimothyEdwards/magic/files/6013856/make.log)

RTimothyEdwards, 2021-02-20:
Don't use -Werror

jay20162016, 2021-02-20:
I don't use -Werror... Maybe it is included in the Makefile?

RTimothyEdwards, 2021-02-20:
Usually what I do in these cases, if the system insists on overly-cautious defaults, is to find the setting in the Makefile (probably under CFLAGS), and remove it.  "-Werror" means "turn all warnings into fatal errors", so removing that one option should allow the compilation to complete.

jay20162016, 2021-02-20:
`CFLAGS                 = -g  -I/usr/X11/include  -I/usr/local/Cellar/tcl-tk/8.6.11_1/include -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.11_1/include`
I don't see -Werror...

jay20162016, 2021-02-20:
It turns out that my default c++ compiler (apple built-in clang) has -Werror by default. Using another compiler worked.

jay20162016, 2021-02-20:
Now I get a segfault....

mooredan, 2021-02-20:
I regularly compile magic on a MacBook Pro and use the attached script to do so.

Current OS version: macOS Catalina 10.15.7
using MacPorts version 2.6.4

Config results in this:

-----------------------------------------------------------
Configuration Summary (principle requirements):

X11:          yes
Python3:      yes
OpenGL:       yes
Vector fonts: yes
Cairo:        yes
Tcl/Tk:       yes
-----------------------------------------------------------

[compile_magic.zip](https://github.com/RTimothyEdwards/magic/files/6016304/compile_magic.zip)

mooredan, 2021-02-20:
hmmm, strange cut and paste formatting

```
X11:          yes
Python3:      yes
OpenGL:       yes
Vector fonts: yes
Cairo:        yes
Tcl/Tk:       yes
```

jay20162016, 2021-02-20:
Yes, I get similar config results, and make and install works without errors. However, upon running magic, I get a segfault.

mooredan, 2021-02-21:
Ugh.  Could be anything, but I guess a bad .so file.  Does the tkcon or main layout window even appear?  I've run magic in gdb once before, but I found it a bit cumbersome (probably because I don't so this very often).  I'll see if I have any notes on how to do it.

How about starting magic like this (just for debugging):

$ /opt/local/bin/magic -noconsole -dnull

Magic 8.3 revision 130 - Compiled on Sat Feb 20 07:22:37 PST 2021.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Using technology "minimum", version 0.0
%

jay20162016, 2021-02-21:
The icon shows up on dock, and I think I've seen the window once or twice. I get the same output as you when I run it:
```
╰─ /usr/local/bin/magic -noconsole -dnull                                                                                   ─╯

Magic 8.3 revision 129 - Compiled on Sat Feb 20 03:15:29 PST 2021.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Using technology "minimum", version 0.0
%
```

mooredan, 2021-02-21:
Well its not totally broke, good. ...how about without the -dnull?  ...or without -noconsole (leaving -dnull in place).  I'm going to guess that you get the crash in both of these cases.  If so, kind of points to the TK/TCL installations.  Does "wish" crash?

jay20162016, 2021-02-21:
`wish`, `magic -dnull` works, however `magic -noconsole` raises a segfault:
```
/usr/local/bin/magic -noconsole                                                                                          ─╯
Use openwrapper to create a new GUI-based layout window
Use closewrapper to remove a new GUI-based layout window

Magic 8.3 revision 129 - Compiled on Sun Feb 21 17:30:36 EST 2021.
Starting magic under Tcl interpreter
Using the terminal as the console.
[1]    81206 segmentation fault  /usr/local/bin/magic -noconsole
```

RTimothyEdwards, 2021-02-21:
@jay20162016 :  FYI, If the compiler declares "-Werror" by default, then you can maybe negate it with "-Wno-error".  But since you have it working with another compiler, then I guess leave well enough alone. . .

The segfault happening only when non-null graphics is selected but not otherwise suggests that you may be linking to a version of Tk that uses Mac graphics.  Magic makes calls directly to X11, so you need a compatible version of Tk that is compiled to X11.  Macports supposedly has one, but I am not a Mac person and don't know much about it.  I am not 100% sure that's the issue, but it sounds like it.

mooredan, 2021-02-21:
My configure command:

./configure --prefix=/opt/local --with-tcl=/opt/local/lib \
--with-tk=/opt/local/lib --x-includes=/opt/local/include --x-libraries=/opt/local/lib

..and this output from configure:

checking for tclConfig.sh... /opt/local/lib/tclConfig.sh
checking for tkConfig.sh... /opt/local/lib/tkConfig.sh
checking for wish executable... /opt/local/bin/wish
checking for tclsh executable... /opt/local/bin/tclsh

Yes, this TK/TCL is coming from MacPorts:

$ port installed | grep -i -e " tk @" -e " tcl @"
  tcl @8.6.10_0+corefoundation+threads (active)
  tk @8.6.10_1+x11 (active)

Maybe the config could detect the OS and ensure that these paths are not coming from MacOS's location.  Homebrew supposedly works too, but I chose the MacPorts path a long time ago.

jay20162016, 2021-02-21:
I use brew's `tck-tk` on `/usr/local`

jay20162016, 2021-02-25:
Also, it seems like brew's `tcl-tk` is compiled without x11 support...

jay20162016, 2021-02-26:
How do you compile `tcl-tk` with x11? I don't think I can get brew to work...

mooredan, 2021-02-26:
Kind of depends upon how much you have invested in homebrew, you could make the jump to macports and not have to compile tk at all.

But here's the github page for the portfile for the tk port, which includes configure options and compiler flags.
https://github.com/macports/macports-ports/blob/master/x11/tk/Portfile

ryandesign, 2021-08-12:
This issue seems to have derailed a bit from the initially-reported problem, which is that magic is calling functions before declaring them. This must be fixed.

Implicit declaration of functions has been prohibited ever since the C99 standard was introduced decades ago, but compilers have allowed it with a warning. The reason it now becomes a problem (in the past year) is that the version of clang included with Xcode 12 and later considers implicit declaration of functions to be an error, not a warning: It defaults to `-Werror=implicit-function-declaration`. Turning this off by specifying `-Werror=no-implicit-function-declaration` is strongly discouraged. Apple made this the default because they introduced new Mac models based on ARM processors, for which they use the brand name Apple Silicon. ARM processors use different calling conventions for variadic and non-variadic functions whereas Intel processors used the same conventions (in most cases) so this was not a problem on Intel. But for ARM the compiler must know before it writes a function invocation's machine code which type of function it is. To know this, it must see the function prototype. So please provide them.

ryandesign, 2021-12-09:
Are you making any progress on declaring your functions before you call them?

just22, 2022-10-29:
After merging PR #191 Magic code compiles with `-Werror=implicit-function-declaration` (which has been added to CFLAG). Tested on a MacBook Air M1 (no more segfaults) and an amd64 arch running OpenBSD.

So I think this issue is resolved and can be closed.

--
Alessandro

RTimothyEdwards, 2022-10-29:
@just22 :  Great, thanks for checking, and thanks again for the hard work getting this done!

closed 2022-10-29 by RTimothyEdwards