RTimothyEdwards/magic #70: ext2spice has problems with instance names that contain a '/'

d-m-bailey, 2021-02-24:
This issue was raised during the extraction of caravel's DFFRAM. Over 150,000 messages were generated.
Example
```
connect(1): no such node COLUMN\[0\].RAMCOLS/B_0_2/WORD\[19\].W/B2/BIT\[6\].FF/Q
connect(1): no such node COLUMN\[0\].RAMCOLS/B_0_2/WORD\[19\].W/B2/BIT\[6\].OBUF/A
connect(1): no such node COLUMN\[0\].RAMCOLS/B_0_1/WORD\[36\].W/B3/BIT\[1\].FF/Q
connect(1): no such node COLUMN\[0\].RAMCOLS/B_0_1/WORD\[36\].W/B3/BIT\[1\].OBUF/A
connect(1): no such node COLUMN\[0\].RAMCOLS/B_0_2/WORD\[22\].W/B2/BIT\[4\].FF/Q
...
```
Mostly fixed with magic 8.3.132
However, caravel/mgmt_core extraction results in 4 messages
```
connect(1): no such node soc.soc_mem.mem.SRAM\/FILLER_32_209/VGND
connect(2): no such node soc.soc_mem.mem.SRAM\/FILLER_32_209/VGND
connect(1): no such node soc.soc_mem.mem.SRAM\/FILLER_173_42/VPWR
connect(2): no such node soc.soc_mem.mem.SRAM\/FILLER_173_42/VPWR
```
The offending lines from mgmt_core.ext appear to be
```
merge "PHY_1162/VPWR" "soc.soc_mem.mem.SRAM/FILLER_173_42/VPWR"
merge "soc.soc_mem.mem.SRAM/FILLER_173_42/VPWR" "PHY_463/VPWR"
merge "soc.soc_mem.mem.SRAM/VGND" "soc.soc_mem.mem.SRAM/FILLER_32_209/VGND"
merge "soc.soc_mem.mem.SRAM/FILLER_32_209/VGND" "PHY_181/VGND"
```
The only place in the file where `FILLER_173_42` and `FILLER_32_209` appear are these 4 lines. It appears that it's trying to merge with an unreferenced internal node from a subcircuit.
And in the resulting spice net, VPWR has been split in 2 (VPWR & `_65235_/VPB`).
```
X_39511_ _39511_/A _64067_/D _65235_/VPB VGND _65235_/VPB VGND sky130_fd_sc_hd__inv_2
XFILLER_188_3270 VGND VPWR VPWR VGND sky130_fd_sc_hd__decap_4
```
To duplicate in `caravel/openlane`, add the following lines to `mgmt_core/config.tcl`
```
set ::env(LVS_EXTRA_GATE_LEVEL_VERILOG) "
       $script_dir/../../verilog/gl/DFFRAM.v
       $script_dir/../../verilog/gl/digital_pll.v"
```
and run
```
flow.tcl -design mgmt_core -tag ext -lvs -net ../verilog/gl/mgmt_core.v -gds ../gds/mgmt_core.gds
```
I've noticed a similar error when extracting `user_project_wrapper` of `caravel_vdp_lite`, but it doesn't appear to cause any connectivity errros:
```
connect(1): no such node mprj\/FILLER_334_206/VGND
connect(1): no such node mprj\/FILLER_301_39/VPWR
connect(1): no such node mprj\/FILLER_367_39/VPWR
```

RTimothyEdwards, 2021-02-24:
I am now changing `/` to `\/` (backslash-escaped) where `/` appears in node and instance names.  But it looks like there are places in the .ext file where there are multiple cells in the hierarchy in the "merge" lines, so they cannot just be considered part of the instance name.  I will probably need to print the backslash escapes into the .ext file rather than try to create them on reading back the .ext file.

RTimothyEdwards, 2021-02-24:
Gave up on that and am now just replacing "/" in instance names read from GDS or DEF to "_".