RTimothyEdwards/magic #71: ext2spice problems with cells names beginning with numbers and net names resembling busses

d-m-bailey, 2021-03-06:
1. In the efabless Google shuttle project `caravel_UNCC_MPW_1`, the following cells begin with numbers.
```
10good
1T-cell
2good
2x2-array
3good
4good
5good
6good
7good
8good
9good
```
`ext2spice` strips the leading numbers, resulting in duplicate subckt definitions.
```
.subckt good VSUBS m3_n30_n637# a_n180_1347# a_n180_n986# Sw-1_0/w_1442_592# m2_240_n362#
.subckt good VSUBS m2_426_2318# 2good_1/a_n180_n986# 2good_0/a_n180_1347# m3_1772_2124#
.subckt good VSUBS Sw-1_0/li_402_584# 3good_1/2good_1/a_n180_n986# m3_2006_4492# m2_444_4830#
.subckt good VSUBS Sw-1_0/li_402_584# 4good_0/3good_0/2good_0/a_n180_1347# 4good_1/3good_1/2good_1/a_n180_n986#
.subckt good VSUBS Sw-1_0/li_402_584# m2_458_19930# 5good_0/4good_0/3good_0/2good_0/a_n180_1347#
.subckt good VSUBS Sw-1_0/li_402_584# 6good_0/5good_0/4good_0/3good_0/2good_0/a_n180_1347#
.subckt good VSUBS 7good_0/6good_0/5good_1/4good_1/3good_1/2good_1/a_n180_n986# Sw-1_0/li_402_584#
.subckt good VSUBS m1_34870_45168# Sw-1_0/li_402_584# 8good_1/7good_0/6good_1/5good_1/4good_1/3good_1/2good_1/a_n180_n986#
.subckt good GND VDD D0 D1 D2 D3 Sw-1_0/w_1442_592# D4 D5 D6 D7 D8 D9
```
2. The `2x2-array.ext` file contains lines that throw an error in `ext2spice`. Looks like it thinks the bus widths differ.
```
2x2-array.ext, line 51: Number of subscripts doesn't match
2x2-array.ext, line 53: Number of subscripts doesn't match
2x2-array.ext, line 55: Number of subscripts doesn't match
2x2-array.ext, line 57: Number of subscripts doesn't match

51 merge "1T-cell_0[0:1,0:0]/body" "body"
53 merge "1T-cell_0[0:1,0:0]/body" "body"
55 merge "1T-cell_0[0:1,0:0]/gate" "gate"
57 merge "1T-cell_0[0:0,0:1]/source" "source"
```

RTimothyEdwards, 2021-03-06:
Leading numbers are illegal SPICE syntax which is why magic is stripping them.  I could probably introduce some kind of setting to allow them, especially for LVS purposes.  I'm not aware of anything syntactically ambiguous about a cell name or net name beginning with a number, and I'm not sure if, say, ngspice would flag it or not.  If not, then two strikes against it and I'll just remove the behavior altogether.

d-m-bailey, 2021-03-06:
Would prepending '_' work?

RTimothyEdwards, 2021-03-07:
It's not at all clear to me why the merge lines complain about subscripts that don't match.  I'm not as familiar with the array handling as other parts of the extraction code.

RTimothyEdwards, 2021-03-07:
Yes, I think it makes more sense to prepend something to the name rather than remove something from the name.  The intent was to deal with something internally generated by magic, not with the expectation that cell names would begin with a number.

RTimothyEdwards, 2021-03-07:
Probably better to make the behavior selectable by command, but for now, anyway, I have modified magic to prepend an "x" rather than remove the non-alphabetical characters (underscore won't do, according to the ngspice manual;  the name has to begin with A-Z. . .  but it's still not clear to me why ngspice would care;  any name would be unambiguous as long as it didn't contain spaces or "=").