RTimothyEdwards/magic #72: run-time error on magic master branch since release 8.3.130

AppMathIsRealMath, 2021-04-06:
Building against master has given me a startup run-time error since the 8.3.130 release. The main screen is blank and the tkcon 2.3 Main console reports:

Magic 8.3 revision 152 - Compiled on Tue 06 Apr 2021 06:11:34 AM MDT.
Starting magic under Tcl interpreter
Using Tk console window
Using TrueColor, VisualID 0x21 depth 24
Processing system .magicrc file
New windows will not have a title caption.
New windows will not have scroll bars.
New windows will not have a border.
error in slave eval:
value(s) for the -zoom option must be positive
Main console display active (Tcl8.6.10 / Tk8.6.10)
% 

I'm running Solus Linux - are my tcl/tk libraries too new or too old?

RTimothyEdwards, 2021-04-06:
There isn't anything relevant in my comments for the 8.3.130 commit.  However, there is something possibly relevant in 8.3.131, where I was using a Tk command to query a font pixel size so I could keep all the sizes of fonts and icons sane when used with a high-resolution display.  Possibly this results in not-so-sane values depending on how fonts are installed.  If you can confirm that rev. 130 works but rev. 131 doesn't, then I can get you to run some Tk commands in the console to see what values it produces, so I could detect this condition and drop back to the original behavior.  Assuming that's what the issue is.

AppMathIsRealMath, 2021-04-06:
Sorry about that, I phrased it incorrectly - 8.3.130 is the last working release for me, 8.3.131 onwards is broken for me.

Sure, let me know what commands you want me to run, I'll be happy to.

RTimothyEdwards, 2021-04-06:
@AppMathIsRealMath :  Try replacing tcltk/wrapper.tcl line 1144:

`set Opts(scale) [expr {int([font measure TkDefaultFont M] / 10)}]`

with 

`set Opts(scale) [expr {max(1, int([font measure TkDefaultFont M] / 10))}]`

I see that if the default font size is less than 10, then Opts(scale) rounds down to zero, which I expect causes all manner of bad things to happen.

AppMathIsRealMath, 2021-04-06:
That fixed it, thanks!

RTimothyEdwards, 2021-04-07:
Good to know.  I'll commit that fix when I finish working on the current problem.

RTimothyEdwards, 2021-04-07:
I have updated the code (on opencircuitdesign.com;  github will be mirrored overnight) and will go ahead and close this issue.  Thanks for the bug report and thanks again for helping to debug it!

closed 2021-04-07 by RTimothyEdwards