RTimothyEdwards/magic #77: Unified WM_CLASS property "magic" for all windows

reikred, 2021-05-20:
Can we get WM_CLASS set to "magic" for all  magic windows? Right now xprop indicates for the console window that

_NET_WM_NAME(UTF8_STRING) = "tkcon 2.3 Main" 
WM_NAME(STRING) = "tkcon 2.3 Main"          <------- a better value would be "magic:console" or perhaps "magic:tkcon"
WM_CLASS(STRING) = "tkcon.tcl", "Tkcon.tcl" <------- should be "magic", "magic"

and for layout windows the values are something like

_NET_WM_NAME(UTF8_STRING) = "layout1"
WM_NAME(STRING) = "layout1"                   <-------- a better value would be magic:layoutfilename (not the full path, you can see that in 
WM_CLASS(STRING) = "layout1", "Toplevel" <-------- should be "magic", "magic"

Having proper WM_CLASS and WM_NAME values makes life easier for the window manager and the user when it comes to finding, sorting and categorizing windows.

As a starting point, one place where one  hack-y fix could be applied is tkcon.tcl:620

#wm title $root "tkcon $PRIV(version) $title"
wm title $root "magic"

I am however not at all a tcl/tk or x11 expert, and an expert could likely do something more intelligent and unified to set WM_CLASS and WM_NAME properly in all windows.

reikred, 2021-05-21:
UPDATE: I have made _some_ progress on this one, more to come.

RTimothyEdwards, 2021-05-22:
I looked at it briefly, but whenever I tried anything it started breaking things and producing error messages.  So if you're making headway, I'll just wait until you have something that works.  If you have any questions about implementation, just ask.