RTimothyEdwards/magic #79: Device menu not appearing in magic with skywater pdk

aakash-ics, 2021-06-06:
Hi, 

I am trying to use the skywater 130nm open-pdk with magic. Seeing some tutorials/videos about using magic and skywater 130nm I realized that I am not able to see the devices menu of magic. 

SO I found out a file sky130A.tcl which seem to be the device generator menu so I tried sourcing it from the console of magic.

I tried opening magic  inside the ~/open_pdks/sky130/sky130A/libs.tech/magic directory using the following statement in terminal

magic -T sky130A.tech sky130A.tcl 

[image]

The above image is my case.

[image]

this image above is one of tutorial that I saw on youtube using magic and sky130.

Also I saw one tutorial from efabless where the person was using magic on the efabless website and there also I could see the devices menu for parameterized device generation.

Kindly help.

Thanks in Advance
Aakash Gupta

RTimothyEdwards, 2021-06-06:
Sourcing the file like that works for any _new_ windows created;  so if you then did Window->New, the new window would have the device menu.  The Tcl script has to be loaded before the window is created, so it's done in the startup file.  Since the startup file does a number of other important things in addition to loading the device generator script, I recommend doing, e.g., "magic -d OGL -rcfile <path_to>/sky130A/libs.tech/magic/sky130A.magicrc".  The startup script loads the correct tech file, so you don't need the "-T" option on the command line, just the "-rcfile" option.

aakash-ics, 2021-06-06:
Thanks  RTimothyEdwards,

It worked !

[image]

closed 2021-06-06 by aakash-ics

RTimothyEdwards, 2021-06-06:
Unrelated to the PDK, but related to the "-d OGL" argument:  To make the icons on the right show up properly, either use "-d XR" instead of "-d OGL" or else configure with the option "--enable-cairo-offscreen" before compiling.

aakash-ics, 2021-06-06:
Thanks RTimothyEdwards for the additional information it looks great now!

[image]