RTimothyEdwards/magic #80: make DESTDIR=xxxx install fails due to tclmagic.so issue?

mithro, 2021-06-12:
```
gcrDebug.c:946:19: warning: implicit declaration of function 'niceabort' is invalid in C99 [-Wimplicit-function-declaration]
--- linking libgcr.o
--- compiling tcltk/tclmagic.o
In file included from tclmagic.c:30:
In file included from tclmagic.c:30:
tclmagic.c:538:9: warning: implicit declaration of function 'mainInitBeforeArgs' is invalid in C99 [-Wimplicit-function-declaration]
tclmagic.c:539:9: warning: implicit declaration of function 'mainDoArgs' is invalid in C99 [-Wimplicit-function-declaration]
tclmagic.c:560:9: warning: implicit declaration of function 'mainInitAfterArgs' is invalid in C99 [-Wimplicit-function-declaration]
tclmagic.c:725:9: warning: implicit declaration of function 'mainInitFinal' is invalid in C99 [-Wimplicit-function-declaration]
tclmagic.c:932:2: warning: implicit declaration of function 'txGetFileCommand' is invalid in C99 [-Wimplicit-function-declaration]
--- linking libtcltk.o
--- compiling magic/magicTop.o
magicTop.c:53:5: warning: implicit declaration of function 'magicMain' is invalid in C99 [-Wimplicit-function-declaration]
--- linking libmagic.o
--- compiling net2ir/net2ir.o
net2ir.c:122:9: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
net2ir.c:122:9: note: place parentheses around the assignment to silence this warning
net2ir.c:122:9: note: use '==' to turn this assignment into an equality comparison
--- linking libnet2ir.o
--- making Tcl shared libraries
make[2]: *** No rule to make target '../extract/libextract.o', needed by 'tclmagic.so'.  Stop.
--- installing executable to /opt/magic/usr/local/bin
--- installing runtime files to /opt/magic/usr/local/lib
In file included from cifin.c:137:
./cifin-ibm.gen:2:1: warning: '/*' within block comment [-Wcomment]
/*  This is prelimanary, please let me know if anything here seems
^
In file included from cifin.c:139:
./cifin-ibm.gen:2:1: warning: '/*' within block comment [-Wcomment]
/*  This is prelimanary, please let me know if anything here seems
^
2 warnings generated.
In file included from cifin.c:84:
./cifin-cmos14b.gen:2:1: warning: '/*' within block comment [-Wcomment]
/*  This is prelimanary, please let me know if anything here seems
^
1 warning generated.
In file included from cifin.c:115:
./cifin-cmos14b.gen:2:1: warning: '/*' within block comment [-Wcomment]
/*  This is prelimanary, please let me know if anything here seems
^
In file included from cifin.c:117:
./cifin-cmos14b.gen:2:1: warning: '/*' within block comment [-Wcomment]
/*  This is prelimanary, please let me know if anything here seems
^
2 warnings generated.
make[2]: *** No rule to make target '../extract/libextract.o', needed by 'tclmagic.so'.  Stop.
make: *** No rule to make target 'FROM'.  Stop.
The command '/bin/sh -c git clone https://github.com/RTimothyEdwards/magic.git /tmp/magic  && mkdir -p /opt/magic/usr/local/bin  && cd /tmp/magic  && ./configure  && make -j $(nproc)  && make DESTDIR=/opt/magic install
````

RTimothyEdwards, 2021-06-12:
This used to work.  I don't know where it got the target "FROM" from. . .

mithro, 2021-06-12:
I think this is actually a missing `tcl-expect` package?

mithro, 2021-06-12:
The `FROM` bit was a mistake in my part.

mithro, 2021-06-12:
`make[2]: *** No rule to make target '../extract/libextract.o', needed by 'tclmagic.so'.  Stop.`

mithro, 2021-06-12:
```
--- compiling extract/ExtCell.o
rm -f ExtCell.o
clang -g -m64 -fPIC -Wimplicit-int -fPIC  -I/usr/include/tcl8.6/tk-private/generic -I/usr/include/tcl8.6 -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DHAVE_LIBGL=1 -DHAVE_LIBGLU=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -DTHREE_D=1 -Dlinux=1 -DSYSV=1 -DISC=1 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DGCORE=\"no\" -DSHDLIB_EXT=\".so\" -DNDEBUG  -c ExtCell.c
In file included from ExtCell.c:33:
In file included from ../database/database.h:41:
In file included from ../bplane/bplane.h:175:
In file included from ../bplane/bpOpaque.h:34:
../utils/ihash.h:53:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
In file included from ExtCell.c:33:
In file included from ../database/database.h:41:
In file included from ../bplane/bplane.h:214:
../bplane/bpEnum.h:509:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextBINS(bpe, 0)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:509:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextBINS(bpe, 0)) return e;
          ^
         (                       )
../bplane/bpEnum.h:509:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextBINS(bpe, 0)) return e;
          ^
          ==
../bplane/bpEnum.h:513:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextBINS(bpe, 1)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:513:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextBINS(bpe, 1)) return e;
          ^
         (                       )
../bplane/bpEnum.h:513:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextBINS(bpe, 1)) return e;
          ^
          ==
../bplane/bpEnum.h:517:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextINBOX(bpe, 0)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:517:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextINBOX(bpe, 0)) return e;
          ^
         (                        )
../bplane/bpEnum.h:517:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextINBOX(bpe, 0)) return e;
          ^
          ==
../bplane/bpEnum.h:521:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextINBOX(bpe, 1)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:521:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextINBOX(bpe, 1)) return e;
          ^
         (                        )
../bplane/bpEnum.h:521:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextINBOX(bpe, 1)) return e;
          ^
          ==
../bplane/bpEnum.h:525:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextHASH(bpe)) return e;
         ~^~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:525:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextHASH(bpe)) return e;
          ^
         (                    )
../bplane/bpEnum.h:525:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextHASH(bpe)) return e;
          ^
          ==
ExtCell.c:115:2: error: non-void function 'ExtCell' should return a value [-Wreturn-type]
[... 7 lines cut ...]
                 ^
            (                            )
ExtCell.c:183:11: note: use '==' to turn this assignment into an equality comparison
        if (endp = strrchr(ends + 1, '.'))
                 ^
                 ==
ExtCell.c:402:31: warning: implicit declaration of function 'extSubtree' is invalid in C99 [-Wimplicit-function-declaration]
    if (!SigInterruptPending) extSubtree(extParentUse, reg, f);
                              ^
ExtCell.c:403:31: warning: implicit declaration of function 'extArray' is invalid in C99 [-Wimplicit-function-declaration]
    if (!SigInterruptPending) extArray(extParentUse, f);
                              ^
ExtCell.c:406:14: warning: implicit declaration of function 'ExtFreeLabRegions' is invalid in C99 [-Wimplicit-function-declaration]
    if (reg) ExtFreeLabRegions((LabRegion *) reg);
             ^
ExtCell.c:407:5: warning: implicit declaration of function 'ExtResetTiles' is invalid in C99 [-Wimplicit-function-declaration]
    ExtResetTiles(def, extUnInit);
    ^
ExtCell.c:411:2: warning: implicit declaration of function 'extLength' is invalid in C99 [-Wimplicit-function-declaration]
        extLength(extParentUse, f);
        ^
12 warnings and 1 error generated.
make[2]: *** [../rules.mak:26: ExtCell.o] Error 1
make[2]: Leaving directory '/tmp/magic/extract'
```

mithro, 2021-06-13:
I think you need;
```diff
diff --git a/utils/ihash.h b/utils/ihash.h
index 9c5d3f3..3efe01c 100644
--- a/utils/ihash.h
+++ b/utils/ihash.h
@@ -50,6 +50,7 @@ static __inline__ int IHashAlignedSize(int size)
     int result;
     /* Expand size to be double-word (64 bit) aligned */
     result = ((size + 7) / 8) * 8;
+    return result;
 }
 
 /* The IHashTable struct should not be manipulated directly by clients */
```

?

mithro, 2021-06-13:
Actually, looks like it is;
```diff
diff --git a/extract/ExtCell.c b/extract/ExtCell.c
index d2289dd..bb470f6 100644
--- a/extract/ExtCell.c
+++ b/extract/ExtCell.c
@@ -112,7 +112,7 @@ ExtCell(def, outName, doLength)
        TxError("Cannot open output file: ");
        perror(filename);
 #endif
-       return;
+       return NULL;
     }
 
     extNumFatal = extNumWarnings = 0;
```

mithro, 2021-06-13:
Actually looks like https://github.com/RTimothyEdwards/magic/pull/75 fixes the problem.

RTimothyEdwards, 2021-06-13:
Indeed, I just missed #75 (too many things going on that day, I guess).

RTimothyEdwards, 2021-06-13:
Now fixed on opencircuitdesign.com master branch.

closed 2021-08-07 by RTimothyEdwards