RTimothyEdwards/magic #82: ext2spice does not wrap the top level devices in a subcircuit, which causes LVS to fail.

d-m-bailey, 2021-06-29:
Device level LVS on some of the submodules of skywater/efabless mpw-1 previously passed, but now the top level of the extracted netlist is not wrapped in a subcircuit definition and fails LVS.

Failing message
`Cannot find cell digital_pll in file /mnt/mpw/1/caravel/openlane/digital_pll/runs/ext/results/magic/digital_pll.gds.spice`

The spice file contains
```
...
* Top level circuit digital_pll

X_294_ VGND _400_/C _564_/D _564_/Q VPWR _400_/C VPWR VGND _388_/B sky130_fd_sc_hd__a2bb2o_4
...
XFILLER_31_111 VGND VPWR VPWR VGND sky130_fd_sc_hd__decap_8
.end
```
Was expecting something like the following
```
...
* Top level circuit digital_pll
.subckt digital_pll VPWR VGND ...
X_294_ VGND _400_/C _564_/D _564_/Q VPWR _400_/C VPWR VGND _388_/B sky130_fd_sc_hd__a2bb2o_4
...
XFILLER_31_111 VGND VPWR VPWR VGND sky130_fd_sc_hd__decap_8
.ends
```
To duplicate this 
```
cd $OPENLANE_HOME
git clone https://github.com/efabless/caravel.git
make mount
cd caravel/openlane
gunzip ../gds/digital_pll.gds.gz
flow.tcl -design digital_pll -tag ext -lvs -net ../verilog/gl/digital_pll.v -gds ../gds/digital_pll.gds
```
This may be a problem with the openlane scripts.

RTimothyEdwards, 2021-06-29:
This should be logged as an issue for openlane, not for magic.  Magic will generate a top level subcircuit wrapper as long as (1) the cell has defined ports, and (2) the "ext2spice subcircuit top on" option is set (which is also set by doing "ext2spice lvs", which is what OpenLane should be doing.  OpenLane is supposed to be doing both of those in the netlist generation scripts.

d-m-bailey, 2021-07-01:
The 'problem' is that openlane does not specify `cif istyle sky130(vendor)` before reading the GDS.

closed 2021-07-01 by d-m-bailey