RTimothyEdwards/magic #86: Segmentation fault (core dumped)

harshithsn, 2021-08-07:
I got segmentation fault (core dumped) error by loading `sky130A.tech` file.
`magic -T sky130A.tech `
I have attached `sky130A.tech` file in below link, Please resolve my issue

[sky130A.zip](https://github.com/RTimothyEdwards/magic/files/6949060/sky130A.zip)

Thank You

RTimothyEdwards, 2021-08-07:
Probably your version of magic is way out of date.

harshithsn, 2021-08-07:
> Probably your version of magic is way out of date.

@RTimothyEdwards sir, are you able to open magic with `sky130A.tech` technology file?

RTimothyEdwards, 2021-08-07:
Yes, I can open magic with your sky130A.tech.  What version of magic are you using?

harshithsn, 2021-08-07:
> Yes, I can open magic with your sky130A.tech. What version of magic are you using?

8.1.198

RTimothyEdwards, 2021-08-07:
Your version of magic is three and a half years out of date so no, it isn't compatible at all with the SkyWater PDK and of course it crashes.

harshithsn, 2021-08-07:
> Your version of magic is three and a half years out of date so no, it isn't compatible at all with the SkyWater PDK and of course it crashes.

@RTimothyEdwards Thank you sir, I just updated magic and now I can easily run magic with `sky130A.tech`

Thank You 😃🙏

closed 2021-08-07 by harshithsn