RTimothyEdwards/magic #92: Segfault setting DRC style

gatecat, 2021-11-05:
Run `magic -T $OPEN_PDKS/sky130/magic/sky130.tech  -dnull -noconsole do_drc.tcl` in the following archive: https://ds0.me/magic_drc_segfault.tar.xz

It segfaults with backtrace:

```
#0  0x00007ffff5bc18f7 in DBSrPaintArea (hintTile=0x0, plane=0x7fffffffc2d8, rect=0x7ffff5dc6340 <TiPlaneRect>, mask=0x7ffff68fce40 <CIFSolidBits>, func=0x7ffff5ccfd50 <cifProcessResetFunc>, arg=0)
    at DBtiles.c:376
#1  0x00007ffff5cdaf26 in CIFGenLayer (op=0x55555c544b70, area=0x7fffffffbe60, cellDef=0x55555b0442a0, origDef=0x55555b0442a0, temps=0x7fffffffbe80, hier=0 '\000', clientdata=0) at CIFgen.c:4737
#2  0x00007ffff5cdbb1c in CIFGen (cellDef=0x55555b0442a0, origDef=0x55555b0442a0, area=0x7fffffffc830, planes=0x7ffff68f9b60 <CIFPlanes>, layers=0x7ffff6672440 <DBAllTypeBits>, replace=1 '\001', 
    genAllPlanes=1 '\001', hier=0 '\000', clientdata=0) at CIFgen.c:5037
#3  0x00007ffff5bd54e4 in drcCifCheck (arg=0x7fffffffc750) at DRCcif.c:554
#4  0x00007ffff5bd146b in DRCBasicCheck (celldef=0x55555b0442a0, checkRect=0x7fffffffc830, clipRect=0x7fffffffc860, function=0x7ffff5bd9f2e <drcListallError>, cdata=140737488341488) at DRCbasic.c:391
#5  0x00007ffff5bdc67e in DRCInteractionCheck (def=0x55555b0442a0, area=0x7fffffffca00, erasebox=0x7fffffffca00, func=0x7ffff5bd9f2e <drcListallError>, cdarg=140737488341488) at DRCsubcell.c:771
#6  0x00007ffff5bda962 in drcWhyAllFunc (scx=0x7fffffffc9f0, cdarg=0x0) at DRCmain.c:633
#7  0x00007ffff5bda6d8 in DRCWhyAll (use=0x5555599707b0, area=0x7fffffffcac0, fout=0x0) at DRCmain.c:551
#8  0x00007ffff5b72a44 in CmdDrc (w=0x5555556558b0, cmd=0x55555bc72700) at CmdCD.c:3946
#9  0x00007ffff5c5ad8b in WindExecute (w=0x5555556558b0, rc=93824993280272, cmd=0x55555bc72700) at windMain.c:415
#10 0x00007ffff5bce22a in DBWcommands (w=0x5555556558b0, cmd=0x55555bc72700) at DBWprocs.c:632
#11 0x00007ffff5c5763f in WindSendCommand (w=0x5555556558b0, cmd=0x55555bc72700, quiet=1 '\001') at windSend.c:263
#12 0x00007ffff5c4f61d in TxTclDispatch (clientData=0x0, argc=3, argv=0x555555593a50, quiet=1 '\001') at txCommands.c:1180
#13 0x00007ffff5d6c820 in _tcl_dispatch (clientData=0x0, interp=0x555555585ba0, argc=3, argv=0x555555593a50) at tclmagic.c:400
#14 0x00007ffff7c908c6 in TclInvokeStringCommand () from /usr/lib/libtcl8.6.so
#15 0x00007ffff7c95c62 in TclNRRunCallbacks () from /usr/lib/libtcl8.6.so
#16 0x00007ffff7c97b3a in TclEvalEx () from /usr/lib/libtcl8.6.so
#17 0x00007ffff7d5236a in Tcl_FSEvalFileEx () from /usr/lib/libtcl8.6.so
#18 0x00007ffff7d5250a in Tcl_EvalFile () from /usr/lib/libtcl8.6.so
#19 0x00007ffff5ca2ea7 in mainInitFinal () at main.c:1182
#20 0x00007ffff5d6d35f in _magic_startup (clientData=0x0, interp=0x555555585ba0, argc=1, argv=0x555555593610) at tclmagic.c:725
#21 0x00007ffff7c908c6 in TclInvokeStringCommand () from /usr/lib/libtcl8.6.so
#22 0x00007ffff7c95c62 in TclNRRunCallbacks () from /usr/lib/libtcl8.6.so
#23 0x00007ffff7c97b3a in TclEvalEx () from /usr/lib/libtcl8.6.so
#24 0x00007ffff7d5236a in Tcl_FSEvalFileEx () from /usr/lib/libtcl8.6.so
#25 0x00007ffff7d5250a in Tcl_EvalFile () from /usr/lib/libtcl8.6.so
#26 0x00007ffff7d59ed5 in Tcl_SourceRCFile () from /usr/lib/libtcl8.6.so
#27 0x00007ffff7d5a4b6 in Tcl_MainEx () from /usr/lib/libtcl8.6.so
#28 0x0000555555555241 in main (argc=7, argv=0x7fffffffd9e8) at magicdnull.c:49
```

seems to be related to setting `drc style drc(full)`

RTimothyEdwards, 2021-11-05:
Of course any segfault is bad, and I will go fix that particular problem, but the underlying issue here is that `$OPEN_PDKS/sky130/magic/sky130.tech` is a *source* file and is not a proper magic tech file, nor is it meant to be.  You need to install the PDK and then run magic on the installed tech file, which would be `$PDK_ROOT/sky130A/libs.tech/magic/sky130A.tech`.  But what's more, you should not be using the `-T` switch on magic, but rather `magic -rcfile $PDK_ROOT/sky130A/libs.tech/magic/sky130A.magicrc`.

gatecat, 2021-11-05:
Oops, thanks very much for the fast response! Using the proper rcfile it works fine, sorry about that.

RTimothyEdwards, 2021-11-05:
No need to be sorry.  Segfaults should always be reported, even for a garbage in/garbage out situation.

RTimothyEdwards, 2021-11-05:
I have fixed the segfault issue (pushed to opencircuitdesign.com;  will update in github overnight).  So I will go ahead and close this issue.

closed 2021-11-05 by RTimothyEdwards