RTimothyEdwards/magic #96: false DRC error for 45 degree path under large squarea of poly

marknmartin, 2021-11-15:
[bent_channel_test.zip](https://github.com/RTimothyEdwards/magic/files/7539612/bent_channel_test.zip)

The attached layout is targeted to the ONSemiC5 process.    There area a long, thin channel that snakes under a sheet of poly.   The DRC is flagging minimum length errors adjacent to the 45 degree bends.   All channel lengths appear to > min channel length.

mooredan, 2021-11-21:
Hi Mark,

Most designs I've seen done in Magic for this process have used the MOSIS Scalable CMOS (SCMOS) design rules. However, at the top of the .mag file you shared, I see "tech amic5x".  Therefore opening this file in magic is not without issues.  Where can the tech file be found?   ...are non-orthogonal geometries even allowed in this process?

Dan

marknmartin, 2021-11-22:
Hello Dan,

The techfile is one that I have but cannot openly distribute due to NDA issues.   Tim Edwards does have a copy (and is the original author from many many years ago).   The process does allow for non- manhatten geometries.  In fact it is required for a few of the specialty devices.   

The issue has been resolved as of V8.2.228

Mark

closed 2021-11-22 by marknmartin