RTimothyEdwards/magic #102: Segfault in DBCellWrite with sky130_fd_sc_hd__decap_6?

mithro, 2021-11-23:
```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f431811f537 in __GI_abort () at abort.c:79
#2  0x00007f4318178768 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f4318286e2d "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007f431817fa5a in malloc_printerr (str=str@entry=0x7f431828505a "free(): invalid pointer") at malloc.c:5347
#4  0x00007f4318180c14 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:4173
#5  0x00007f43172a50ce in freeMagic (cp=0x555fe0bae730) at malloc.c:146
#6  0x00007f43171b6240 in DBCellWrite (cellDef=0x555fe0db7790, fileName=0x555fe0bafea0 "sky130_fd_sc_hd__decap_6") at DBio.c:3317
#7  0x00007f431717bd78 in cmdSaveCell (cellDef=0x555fe0db7790, newName=0x0, noninteractive=1 '\001', tryRename=1 '\001') at CmdSubrs.c:641
#8  0x00007f431719d2f6 in cmdWriteallFunc (def=0x555fe0db7790, cmd=0x555fe0137da0) at CmdTZ.c:1909
#9  0x00007f43171a7996 in DBCellSrDefs (pattern=0, func=0x7f431719d0e4 <cmdWriteallFunc>, cdata=0x555fe0137da0) at DBcellname.c:2072
#10 0x00007f431719d0d8 in CmdWriteall (w=0x555fdfc68d80, cmd=0x555fe0137da0) at CmdTZ.c:1831
#11 0x00007f4317260eb9 in WindExecute (w=0x555fdfc68d80, rc=93870264633280, cmd=0x555fe0137da0) at windMain.c:415
#12 0x00007f43171dceb9 in DBWcommands (w=0x555fdfc68d80, cmd=0x555fe0137da0) at DBWprocs.c:632
#13 0x00007f431725da75 in WindSendCommand (w=0x555fdfc68d80, cmd=0x555fe0137da0, quiet=1 '\001') at windSend.c:263
#14 0x00007f4317255f5e in TxTclDispatch (clientData=0x0, argc=2, argv=0x555fdfba0800, quiet=1 '\001') at txCommands.c:1180
#15 0x00007f43173640d9 in _tcl_dispatch (clientData=0x0, interp=0x555fdfb93120, argc=2, argv=0x555fdfba0800) at tclmagic.c:400
#16 0x00007f43182ff645 in TclInvokeStringCommand () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#17 0x00007f43183014a2 in TclNRRunCallbacks () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#18 0x00007f43183a129c in Tcl_RecordAndEvalObj () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#19 0x00007f43183c6d99 in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#20 0x0000555fdf05e22d in main (argc=4, argv=0x7ffcc3a5f448) at magicdnull.c:49
```

mithro, 2021-11-23:
I got this backtrace from the core dump found in the built sky130A output @ https://foss-eda-tools.googlesource.com/skywater-pdk/output and the built magic.tar.xz from the associated build output (it's linked from the commit message).

I use the following script to get a Docker container with the debug symbols;
```sh
#!/usr/bin/env bash
# Copyright 2021 SkyWater PDK Authors
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# SPDX-License-Identifier: Apache-2.0

CALLED=$_
[[ "${BASH_SOURCE[0]}" != "${0}" ]] && SOURCED=1 || SOURCED=0

SCRIPT_SRC="$(realpath ${BASH_SOURCE[0]})"
SCRIPT_DIR="$(dirname "${SCRIPT_SRC}")"

SCRIPT_DIR_REL="$(realpath $SCRIPT_DIR --relative-to=$PWD)"

TOP_DIR=$PWD
export TOP_DIR

set -xe

# Cleanup any existing docker build
docker stop --time 0 builder || true
docker rm builder || true

# Start the docker container to do the build inside.
docker run -dt \
  --mount type=bind,source="$(pwd)",target=/host \
  --name builder \
  debian || docker start builder

docker ps

DOCKER_CMD="docker exec builder"

$DOCKER_CMD ls /host

cat > debug.list <<EOF
deb http://deb.debian.org/debian-debug/ bullseye-debug main
deb http://deb.debian.org/debian-debug/ bullseye-proposed-updates-debug main
EOF

$DOCKER_CMD mv /host/debug.list /etc/apt/sources.list.d/
$DOCKER_CMD chown root:root /etc/apt/sources.list.d/debug.list

$DOCKER_CMD apt-get update -qq

$DOCKER_CMD apt-cache search tcsh dbg

$DOCKER_CMD apt-get -y install --no-install-recommends \
    ca-certificates \
    curl \
    wget \
    python3 \

$DOCKER_CMD apt-get -y install --no-install-recommends \
    build-essential \
    gdb \
    git \
    make \

$DOCKER_CMD apt-get -y install --no-install-recommends \
    csh \
    libcairo2-dbgsym \
    libglu1-mesa-dbgsym \
    libncurses6-dbgsym \
    libpython3-all-dbg \
    libx11-6-dbgsym \
    m4 \
    python3.9-dbg \
    tk8.6-dbgsym \
    tcl8.6-dbgsym \
    tcl-expect-dbgsym \
    tcsh-dbgsym \
    libc6-dbg \
```

And then ran `docker exec -it builder gdb /host/out/magic/lib/magic/tcl/magicdnull /host/out/core`

mithro, 2021-11-23:
The other core dump seems to be in a similar place?
```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007fa752576537 in __GI_abort () at abort.c:79
#2  0x00007fa7525cf768 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fa7526dde2d "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007fa7525d6a5a in malloc_printerr (str=str@entry=0x7fa7526dc05a "free(): invalid pointer") at malloc.c:5347
#4  0x00007fa7525d7c14 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:4173
#5  0x00007fa7516fc0ce in freeMagic (cp=0x559c00a98970) at malloc.c:146
#6  0x00007fa75160d240 in DBCellWrite (cellDef=0x559c000f1d50, fileName=0x559c00aeffc0 "sky130_fd_io__overlay_vccd_hvc") at DBio.c:3317
#7  0x00007fa7515d2d78 in cmdSaveCell (cellDef=0x559c000f1d50, newName=0x0, noninteractive=1 '\001', tryRename=1 '\001') at CmdSubrs.c:641
#8  0x00007fa7515f42f6 in cmdWriteallFunc (def=0x559c000f1d50, cmd=0x559c005a7740) at CmdTZ.c:1909
#9  0x00007fa7515fe996 in DBCellSrDefs (pattern=0, func=0x7fa7515f40e4 <cmdWriteallFunc>, cdata=0x559c005a7740) at DBcellname.c:2072
#10 0x00007fa7515f40d8 in CmdWriteall (w=0x559bfeb6bd80, cmd=0x559c005a7740) at CmdTZ.c:1831
#11 0x00007fa7516b7eb9 in WindExecute (w=0x559bfeb6bd80, rc=94128481728448, cmd=0x559c005a7740) at windMain.c:415
#12 0x00007fa751633eb9 in DBWcommands (w=0x559bfeb6bd80, cmd=0x559c005a7740) at DBWprocs.c:632
#13 0x00007fa7516b4a75 in WindSendCommand (w=0x559bfeb6bd80, cmd=0x559c005a7740, quiet=1 '\001') at windSend.c:263
#14 0x00007fa7516acf5e in TxTclDispatch (clientData=0x0, argc=2, argv=0x559bfeaa3800, quiet=1 '\001') at txCommands.c:1180
#15 0x00007fa7517bb0d9 in _tcl_dispatch (clientData=0x0, interp=0x559bfea96120, argc=2, argv=0x559bfeaa3800) at tclmagic.c:400
#16 0x00007fa752756645 in TclInvokeStringCommand () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#17 0x00007fa7527584a2 in TclNRRunCallbacks () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#18 0x00007fa7527f829c in Tcl_RecordAndEvalObj () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#19 0x00007fa75281dd99 in Tcl_MainEx () from /usr/lib/x86_64-linux-gnu/libtcl8.6.so
#20 0x0000559bfe18c22d in main (argc=4, argv=0x7ffe34c1f4d8) at magicdnull.c:49
(gdb) 
```

RTimothyEdwards, 2021-11-23:
Because of magic's weird one-off allocation, it's really hard to debug these problems without either having a reproducible example, or running it through valgrind.