RTimothyEdwards/magic #107: port count mismatch in extracted spice netlist

d-m-bailey, 2021-12-03:
Found these discrepancies in the mpw-2 extracted data.
slot-029: instance `Xres_amp_sync_v2_0` has 2 `avss1p8` ports, subckt `res_amp_sync_v2` has 1.
``` 
 .subckt res_amp_sync_v2 clkp clk_amp clkn vss avdd1p8 rst
 ...
 Xres_amp_sync_v2_0 res_amp_sync_v2_0/clkp res_amp_lin_prog_0/clk clkn avss1p8 avss1p8
 + avdd1p8 res_amp_sync_v2_0/rst res_amp_sync_v2
```
From magic output there's
```
Warning:  Ports "avss1p8_uq1" and "vss" are electrically shorted.
```
I'm wondering if it has something to do with merging ports that have the `_uq*` suffix.

The again in slot-040, instance `Xldo_9_0` (all ldo_* instances) has 2 `ldo_9_0/VREG` ports, subckt `ldo_9` has 1.
```
 .subckt ldo_9 VDD cmp_out std_pt_in_cnt[3] trim3 trim1 std_pt_in_cnt[5] ctrl_out[0]
 ...
 + trim10 trim9 trim8 trim7 trim6 trim5 trim4 reset clk trim2 VREG VSS w_36508_19395#
 ...
 Xldo_9_0 VDD ldo_9_0/cmp_out ldo_9_0/std_pt_in_cnt[3] ldo_9_0/trim3 ldo_9_0/trim1
 ...
 + ldo_9_0/trim4 ldo_9_0/reset clk ldo_9_0/trim2 ldo_9_0/VREG ldo_9_0/VREG VSS VSS
```
From magic output there's
```
Warning:  Ports "VREG:_uq35" and "VREG" are electrically shorted.
```
So maybe the ':' needs to be considered also.

Should be able to replicate the error by downloading the `run_ext.txt` script and 
```
git clone git clone https://foss-eda-tools.googlesource.com/third_party/shuttle/mpw-two/slot-029
cd slot-029
git checkout main
cd gds
gunzip user_analog_project_wrapper.gds.gz
(download the run_ext.sh script)
./run_ext.txt user_analog_project_wrapper.gds res_amp_top
```
[run_ext.txt](https://github.com/RTimothyEdwards/magic/files/7645880/run_ext.txt)

RTimothyEdwards, 2021-12-03:
I assume this was done using the fix for issue #106 ?

d-m-bailey, 2021-12-03:
Sorry @RTimothyEdwards , I should have been more specific. `Magic 8.3 revision 233`
The latest revision of magic hasn't been updated on `https://github.com/rtimothyedwards/magic` yet, but as soon as it is, I'll test it.

RTimothyEdwards, 2021-12-03:
It could be related because #106 exhibited behavior where implicit substrate connections had mismatching ports between the subcircuit definition and the subcircuit instance.  It should be easy enough for me to run your example with the updated magic and see if the problem still occurs or not.

d-m-bailey, 2021-12-13:
The pin order is different, but I'm still seeing this error in `magic 8.3.238`. '*' added for emphasis.
```
.subckt res_amp_sync_v2 rst clkp clk_amp clkn vss avdd1p8
...
Xres_amp_sync_v2_0 res_amp_sync_v2_0/rst res_amp_sync_v2_0/clkp res_amp_lin_prog_0/clk
+ clkn *avss1p8 avss1p8* avdd1p8 res_amp_sync_v2
```

RTimothyEdwards, 2021-12-13:
@d-m-bailey :  It appears that the enumerations used by topVisit() for generating ".subckt" entries, and subcktVisit() for generating "X" entries, are not exactly the same.  I thought they were equivalent, but apparently they differ when there are multiple port names assigned to the same net.  I modified subcktVisit() to use exactly the same enumeration as topVisit().  It is slightly less efficient, but I don't think it will make any noticeable difference in runtime.  Now, with version 8.3.241, they should be guaranteed to give the same result with the same number of port entries.