RTimothyEdwards/magic #108: Incorrect GDS extraction for big designs

stevvik, 2021-12-03:
Hello, Timothy!

I faced with some Magic issue during netlist extraction from GDS for further LVS run.
The problem is that sometimes netlist is extracted from GDS incorrectly.
For example, multiple identical serial launches of extraction gives about 5% incorrect netlists. Other 95% netlists are identical.
I see such Magic behavior only at quite big designs.

I prered testcase with Magic serial extraction run (5 times in cycle) and got incorrect netlist from 3rd attempt.

Incorrectness can be detected by LVS, but I just see that 1 of 5 generated netlists have different file size:

-rw-r--r-- 1 stolbikov users 8597483 Dec  3 13:04 ram_dp_bist.gds.spice.1
-rw-r--r-- 1 stolbikov users 8597483 Dec  3 13:07 ram_dp_bist.gds.spice.2
-rw-r--r-- 1 stolbikov users 8597483 Dec  3 13:11 ram_dp_bist.gds.spice.3
-rw-r--r-- 1 stolbikov users 8597483 Dec  3 13:14 ram_dp_bist.gds.spice.4
-rw-r--r-- 1 stolbikov users **9079459** Dec  3 13:17 ram_dp_bist.gds.spice.5

I created self-sufficient tescase dir that includes gds, runfile all required files and results.
You have to fix path to magic only to run this testcase on your side.

You can find testcase in attached tarball.

GDS is created by OpenLane flow for Sky130 technology. I first faced this issue during LVS in OpenLane.
[MAGIC_EXTRACTION_ISSUE_TESTCASE.tar.zip](https://github.com/RTimothyEdwards/magic/files/7648910/MAGIC_EXTRACTION_ISSUE_TESTCASE.tar.zip)

So, it would be better to update OpenLane with fixed Magic version :)

Thank you in advance!

Eugene Stolbikov

RTimothyEdwards, 2021-12-03:
Thanks for the detailed example.  Since it is an intermittent error, I'm not sure how long it will take me to reproduce the issue.  If it is a memory error (likely) then it should show up with valgrind.  Clearly the pin VPWR got separated from the net.  Why that happened is the big question.

RTimothyEdwards, 2021-12-04:
Instead of a loop, I rewrote the script to keep running until it hits an error, and it ran 91 times successfully overnight, but it did fail in the end.  So it is a repeatable error.  Unfortunately I didn't capture the .ext file, which I need to debug, so I'll have to run again today, but I think I have what I need to debug this.

I did try running with valgrind, which came up with nothing.  My guess is that because some of the .ext file output is enumerated from a hash table, the order of the output lines is somewhat randomized, which suggests that the problem might be a dependence on the order of, say, the "merge" statements.  If so, then the error is in "ext2spice" and should be 100% repeatable when starting from the same .ext file.

stevvik, 2021-12-06:
Timothy,
my investigations refutes your last conclusion!
I found that extraction is always the same, even if spice netlist after ext2spice is incorrect.
And if you run ext2spice twice for one extraction, the netlist after first ext2spice can be corrupted, but after second ext2spice run it is always correct!
It looks like the first ext2spice starts running without waiting for the extraction to complete (big netlist file doesn't have time to close) and incorrect spice netlist is generated, but extraction file is finished writing by the start on second ext2spice run.

RTimothyEdwards, 2021-12-06:
Thanks for the update!  Unfortunately, that result makes it a lot harder to debug.  I was hoping I could get an incorrect .ext file and use it to debug.  I ran it over the weekend and it gave me a correct results 265 times before failing.  The good news, though, is that "ext2spice" runs a lot faster than "extract", so running only "ext2spice" hundreds of times until it hits an error should take only an hour at most, not several days.

stevvik, 2021-12-06:
I think, that multiple ext2spice run will not show you a problem!
One more time: the only FIRST ext2spice gives incorrect spice netlist because it runs immediately after extraction! Any other runs of ext2spice are too late, because extraction file writing has already finished at that time!

RTimothyEdwards, 2021-12-06:
Well, my investigation just refuted your last conclusion.  : )   I ran ext2spice on the same .ext file and it failed on the 24th try.  Fortunately, running ext2spice by itself without loading the design first takes only 6 seconds.

stevvik, 2021-12-06:
That's encouraging! :)
Precisely because ext2spice is runing fast, I run it thousand times and have never got incorrect netlists starting from second run :)

RTimothyEdwards, 2021-12-06:
There might be a difference between running ext2spice multiple times from the same process and running ext2spice from multiple processes of magic.  I am currently doing the 2nd of those, running magic multiple times but calling ext2spice only.

stevvik, 2021-12-06:
I run ext2spice by the first method in the loop inside magic.tcl scrip.
Hope now you will find and fix this bug! :)

RTimothyEdwards, 2021-12-14:
@stevvik :  FYI, I had to debug a memory leak that prevented me from doing continuous runs of "ext2spice" from the same process without running out of memory.  That's been resolved, and I'm now back to trying to track down the issue with the output changing.  It has been running over an hour and has not hit an error yet---suggesting that the error is dependent on having a new process start up.  I don't have any good explanation for why that might happen, so I'll keep it running for a while.

RTimothyEdwards, 2021-12-22:
@stevvik :  I got back to testing and I now cannot get magic to generate an error.  I'm not sure if one of the fixes I made recently also fixed this issue;  I have no particular reason to think so.  If you have a chance to try it again with the most recent version of magic, please let me know if the error still occurs for you.