RTimothyEdwards/magic #109: Missing ports in sram memory array

d-m-bailey, 2021-12-08:
This error has been reported elsewhere, but since I think you've been working on related code, I'll bring it up here again.

The Openram memory array `sky130_sram_*_bitcell_array` layout has 64 distinct nets all labeled `vdd`. magic correctly extracts these as `vdd` and 63 `vdd_uqxxxx` nets. All these met1 nets are connected at a higher level in the layout (with via and met2) .

[image]
[image]

However, magic only creates one `vdd` and 36 `vdd_uqxxx` ports. 
The following 27 `vdd_uqxxxx` nets do not have an associated port.
```
vdd_uq1854
vdd_uq1918
vdd_uq1982
vdd_uq2110
vdd_uq2174
vdd_uq2238
vdd_uq2302
vdd_uq2366
vdd_uq2430
vdd_uq2558
vdd_uq2622
vdd_uq2686
vdd_uq2750
vdd_uq2814
vdd_uq3006
vdd_uq3070
vdd_uq3134
vdd_uq3454
vdd_uq3518
vdd_uq3582
vdd_uq3646
vdd_uq3774
vdd_uq3838
vdd_uq3902
vdd_uq3966
vdd_uq4030
vdd_uq4094
```

You can use the following command to find the coordinates of the cells with unconnected nets.
`grep vdd_uq1854 sky130_sram_1kbyte_1rw1r_32x256_8.spice | sed -e 's/.//' -e 's/ .*//' | grep -f - -w sky130_sram_1kbyte_1rw1r_32x256_8_bitcell_array.ext | grep use`

The attached sample data is from `sky130_sram_1kbyte_1rw1r_32x256_8`.

[bitcell_array.tar.gz](https://github.com/RTimothyEdwards/magic/files/7674242/bitcell_array.tar.gz)

Do the following to recreate the spice file.

- Extract in an empty directory
- ./run_ext.sh sky130_sram_1kbyte_1rw1r_32x256_8

d-m-bailey, 2021-12-08:
magic 8.3.238

RTimothyEdwards, 2021-12-13:
@d-m-bailey :  I have tracked this down to the issue that the bit field in the label record for port number allows a maximum of 16384 ports.  Because of the way the cells are stitched together in the SRAM bitcell array, there are, including both power and ground, more than this number of unique ports in the cell.  Ultimately, the bit field fills up, then the "unique" routine keeps finding the highest numbered port is the maximum number, and sets the new port to 1 + that number, which then in the bit field is zero, so all ports after 16384 have the port number zero and are effectively the same port.  It pains me to have to expand the label record to allow, say, 32 bit records just for the port when most cells only have a handful of declared ports, just to satisfy a very pathological case (which could easily be rendered non-pathological just by making the vertical connections one continuous line instead of chopping them up into many unconnected short segments in the parent cell).  I will think about whether there is a good way to encode port numbers efficiently.

RTimothyEdwards, 2021-12-13:
@d-m-bailey :  I just pushed version 8.3.240, which splits the port number of a label into its own 32-bit record.  This appears to have fixed the problem above, producing a netlist with 64 unique "vdd" ports in the bitcell array.  This also fixed a problem with selection on the same nets (because the selection would include all the ports that overran the bit field and ended up with an index of zero).  Note that the layout remains a pathological case in the sense that labels are poorly represented in magic by linked lists, and certain operations have to scan through all of them.  There is a double-loop in "extract unique" that needs to be got rid of (and is pretty easy to get rid of) that makes "extract unique" run rather long on this layout (although still only a fraction of the time to do the full extraction).

d-m-bailey, 2021-12-13:
Excellent! I'll rerun the mpw-2 designs once the update gets merged. I'm reviewing the 8.3.238 results now.

RTimothyEdwards, 2021-12-14:
@d-m-bailey :  Let me know if this is resolved and can be closed.

d-m-bailey, 2021-12-14:
Looks good on mpw-2's slot-005 and slot-008. Currently checking the storage module of caravel.
LVS is not passing but that may be because of nfet back bias that may be fixed in 8.3.241.

I'll let you know.

d-m-bailey, 2021-12-14:
The cells in the `logic_control_r` block of the srams in the `storage` macro have been flattened resulting in 
```
.subckt control_logic_r D ON S_uq0 S_uq1 S_uq2 S_uq3 D_uq0 D_uq1 D_uq2 D_uq3 csb G
+ wl_en Z_uq15 s_en Z_uq25 Z_uq26 Z_uq27 Z_uq28 Z_uq29 Z_uq30 Z_uq31 Z_uq32 Z_uq34
+ Z_uq35 Z_uq36 Z_uq37 Z_uq39 Z_uq40 Z_uq41 Z_uq42 Z_uq44 Z_uq45 Z_uq46 Z_uq47 Z_uq49
+ Z_uq50 Z_uq51 Z_uq52 Z_uq54 Z_uq55 Z_uq56 Z_uq57 Z_uq59 Z_uq60 Z_uq61 Z_uq62 Z_uq64
+ Z_uq65 Z_uq66 Z_uq67 Z Z_uq2 Z_uq3 Z_uq4 Q Z_uq5 Z_uq6 Z_uq7 clk Z_uq10 Z_uq11 Z_uq12
+ Z_uq13 Z_uq14 Z_uq16 Z_uq17 Z_uq18 S_uq6 Z_uq20 S_uq7 B in Z_uq33 Z_uq38 Z_uq43
+ Z_uq48 Z_uq53 Z_uq58 Z_uq63 Z_uq68 Z_uq0 G_uq4 C out S S_uq16 S_uq15 S_uq14 S_uq13
+ S_uq12 S_uq11 vdd S_uq10 S_uq8
```
and
```
Xcontrol_logic_r_0 gnd control_logic_r_0/D control_logic_r_0/ON control_logic_r_0/S_uq0
+ control_logic_r_0/S_uq1 control_logic_r_0/S_uq2 control_logic_r_0/S_uq3 control_logic_r_0/D_uq0
+ control_logic_r_0/D_uq1 control_logic_r_0/D_uq2 control_logic_r_0/D_uq3 csb1 vdd
+ control_logic_r_0/G bank_0/wl_en1 bank_0/p_en_bar1 bank_0/s_en1 control_logic_r_0/Z_uq25
+ control_logic_r_0/Z_uq26 control_logic_r_0/Z_uq27 control_logic_r_0/Z_uq28 control_logic_r_0/Z_uq29
+ control_logic_r_0/Z_uq30 control_logic_r_0/Z_uq31 control_logic_r_0/Z_uq32 control_logic_r_0/Z_uq34
+ control_logic_r_0/Z_uq35 control_logic_r_0/Z_uq36 control_logic_r_0/Z_uq37 control_logic_r_0/Z_uq39
+ control_logic_r_0/Z_uq40 control_logic_r_0/Z_uq41 control_logic_r_0/Z_uq42 control_logic_r_0/Z_uq44
+ control_logic_r_0/Z_uq45 control_logic_r_0/Z_uq46 control_logic_r_0/Z_uq47 control_logic_r_0/Z_uq49
+ control_logic_r_0/Z_uq50 control_logic_r_0/Z_uq51 control_logic_r_0/Z_uq52 control_logic_r_0/Z_uq54
+ control_logic_r_0/Z_uq55 control_logic_r_0/Z_uq56 control_logic_r_0/Z_uq57 control_logic_r_0/Z_uq59
+ control_logic_r_0/Z_uq60 control_logic_r_0/Z_uq61 control_logic_r_0/Z_uq62 control_logic_r_0/Z_uq64
+ control_logic_r_0/Z_uq65 control_logic_r_0/Z_uq66 control_logic_r_0/Z_uq67 control_logic_r_0/Z
+ control_logic_r_0/Z_uq2 control_logic_r_0/Z_uq3 control_logic_r_0/Z_uq4 control_logic_r_0/Q
+ control_logic_r_0/Z_uq5 control_logic_r_0/Z_uq6 control_logic_r_0/Z_uq7 row_addr_dff_0/clk
+ control_logic_r_0/Z_uq10 control_logic_r_0/Z_uq11 control_logic_r_0/Z_uq12 control_logic_r_0/Z_uq13
+ control_logic_r_0/Z_uq14 control_logic_r_0/Z_uq16 control_logic_r_0/Z_uq17 control_logic_r_0/Z_uq18
+ control_logic_r_0/S_uq6 control_logic_r_0/Z_uq20 control_logic_r_0/S_uq7 control_logic_r_0/B
+ rbl_bl1 control_logic_r_0/Z_uq33 control_logic_r_0/Z_uq38 control_logic_r_0/Z_uq43
+ control_logic_r_0/Z_uq48 control_logic_r_0/Z_uq53 control_logic_r_0/Z_uq58 control_logic_r_0/Z_uq63
+ control_logic_r_0/Z_uq68 control_logic_r_0/Z_uq0 clk1 control_logic_r_0/C control_logic_r_0/out
+ vdd vdd gnd vdd vdd vdd vdd vdd vdd vdd vdd vdd vdd vdd vdd vdd vdd vdd control_logic_r
```
The instance call has 10 more nets than the subcircuit definition. I'll re-extract once `8.3.241` becomes available.

On the bright side, the vdd nets in the memory array look good.