RTimothyEdwards/magic #121: Track touching LEF obstruction causes extraction error

antonblanchard, 2022-01-11:
I'm seeing the following error during extraction:

```
feedback add "Illegal overlap between obsm2 and m2 (types do not connect)" medium
```

It looks like OpenROAD detailed routing has placed a track right up against an obstruction in a macro LEF. I'm a bit lost as to who is to blame:

- Should detailed routing enforce a clearance to a LEF obstruction? (If so how much clearance?)
- Should magic recognise that the obstruction is not metal and allow us to touch (but not overlap)?

As a sanity check, I ran the resulting GDS through extraction, and didn't see these errors.

RTimothyEdwards, 2022-01-11:
I have seen obstruction layers done several ways (suggesting that OpenROAD should have configuration settings to choose how to handle obstruction layers), but by far the most common is to treat obstruction just like metal.  Which means to enforce the spacing between any route and obstruction, and to enforce the wide spacing rule(s) if the obstruction is wide.

So, yes, detailed routing should enforce a clearance to a LEF obstruction, and that clearance should match the metal spacing rules.

Depending on how the LEF views are created, they often are made by just dropping an obstruction area on top of the whole cell and then creating cut-outs for the pins.  That means there is often more obstruction than necessary, such that if you ran a wire too close to the obstruction, then you might very well not create a real violation.  But that's luck, not automation.

antonblanchard, 2022-01-11:
Thanks @RTimothyEdwards, I'll package up a test case and follow up with the OpenROAD team.

antonblanchard, 2022-01-11:
While looking through OpenROAD, I found the LEF `USEMINSPACING OBS OFF ;` which the detailed router looks to obey:

```
Spacing is computed to MACRO OBS shapes as if they were actual routing shapes.
A wide OBS shape would use wide wire spacing rules, and a thin OBS shapes would
use thin wire spacing rules
```

Perhaps we should add that to the LEFs magic writes.

RTimothyEdwards, 2022-01-11:
The LEF/DEF reference says "You cannot mix both types of LEF abstract at the same time", which suggests that generating the statement for each macro is not proper;  the statement should be in the technology LEF file so that it applies to all macros.  The LEF/DEF reference also says that the default behavior is "USEMINSPACING OBS ON", which could explain all the errors we've been seeing (I have asked a number of people to look into this problem, but you're the first person to actually find something relevant, so thanks!).