RTimothyEdwards/magic #125: Substrate/nwell connectivity not extracting correctly in hierarchy

d-m-bailey, 2022-01-25:
magic 8.3.260

mpw-3 caravel chip_io. 

By changing resistor and diode recognition layers and flattening the subcells (`chip_io_5.gds`) and making some changes to the netlists, LVS passes for all the chip_io pads. However, even adding `isosub` to totally enclose each of the io cells, the substrates (including some pwell inside dnwell) are merged at the chip_io level.

The reference directory in the tarfile contains the results I got.

To duplicate, 
1. `tar xzf substrate-issue.tar.gz`
2. `cd substrate-issue`
3. `./run_test`

To find the *pins* that have been merged
`awk -f merge_short.awk chip_io.ext`
You should get 
` "vssd1" "vssa1" "vssio" "vssd2" "vssa2" "vssd" "vssa"`

Here are the pwell inside dnwell nets that are also merged with the above ground nets.
These nets correspond to `mid` and `mid1` of `sky130_fd_io__gpiov2_amux_switch`. In the extracted layout, these nets are `w_9674_19062#`, `w_12765_19062#`, `w_12765_16948#`, and `w_9674_16846#`. The shorted nets have 528 connection count.
```
Instance: sky130_ef_io__gpiov2_pad_wrapped:mprj_pads.area2_io_pad\[17\]           |Instance: \mprj_pads.area1_io_pad[0]
...
  ENABLE_VSWITCH_H = 528                                                          |  ENABLE_VSWITCH_H = 98
...
  VSSIO = 528                                                                     |  VSSIO = 98
  VCCD = 177                                                                      |  VCCD = 143
  VSSD = 528                                                                      |  VSSD = 94
  VSSA = 528                                                                      |  VSSA = 23
  VDDIO_Q = 63                                                                    |  VDDIO_Q = 47
  w_9674_19062# = 528                                                             |  proxyw_9674_19062# = 1
  VCCHIB = 177                                                                    |  VCCHIB = 143
  VSSIO_Q = 528                                                                   |  VSSIO_Q = 47
  w_12765_19062# = 528                                                            |  proxyw_12765_19062# = 1
  VDDA = 25                                                                       |  VDDA = 23
  w_12765_16948# = 528                                                            |  proxyw_12765_16948# = 1
  w_9674_16846# = 528                                                             |  proxyw_9674_16846# = 1
```

[image]

d-m-bailey, 2022-01-25:
[substrate-issue.tar.gz](https://github.com/RTimothyEdwards/magic/files/7930150/substrate-issue.tar.gz)

RTimothyEdwards, 2022-01-25:
The `isosub` layer can be a bit of a pain, because it needs to be in multiple subcells of the hierarchy, and it needs to match up exactly from top down.  But I'll take a look at this.

d-m-bailey, 2022-01-25:
Thanks, Tim. I didn't realize that the `isosub` layer had to be on the top layer too. I'll add it and retry.

I may be wrong, but it seems to me that the top level extract is where the problem occurs (and not ext2spice).

d-m-bailey, 2022-01-28:
@RTimothyEdwards I added `isosub` in the top cell `chip_io` exactly as I added it in each of the IO pad cells, but got unexpected results. magic 8.3.260. Now the IO pad cells don't pass LVS either.

How does substrate extraction work in magic? It seems that if there are no substrate connections, `VSUBS` is assigned. Is the substrate connection then passed from abutting pwells in the parent hierarchy?