RTimothyEdwards/magic #128: Problem installing Magic version 8.3.111 on Ubuntu

jarvis-lorry, 2022-02-07:
I'm attempting a clean installation of Magic version 8.3.111 on Ubuntu but cannot get past the make operation without warnings about undefined references.  "Make install" fails immediately afterwards.

Perhaps my version of gcc is bad.
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

From make.log:
/usr/bin/ld: ../extflat/libextflat.o: in function `efReadDef':
/home/lekan/magic-8.3.111/extflat/EFread.c:200: undefined reference to `DBCellLookDef'
/usr/bin/ld: /home/lekan/magic-8.3.111/extflat/EFread.c:272: undefined reference to `ExtCurStyle'
/usr/bin/ld: /home/lekan/magic-8.3.111/extflat/EFread.c:274: undefined reference to `ExtCurStyle'
/usr/bin/ld: /home/lekan/magic-8.3.111/extflat/EFread.c:275: undefined reference to `ExtCurStyle'
/usr/bin/ld: ../extflat/libextflat.o: in function `CmdAntennaCheck':
/home/lekan/magic-8.3.111/extflat/EFantenna.c:176: undefined reference to `ExtCurStyle'
/usr/bin/ld: /home/lekan/magic-8.3.111/extflat/EFantenna.c:201: undefined reference to `DBWclientID'
/usr/bin/ld: /home/lekan/magic-8.3.111/extflat/EFantenna.c:201: undefined reference to `windCheckOnlyWindow'
/usr/bin/ld: /home/lekan/magic-8.3.111/extflat/EFantenna.c:234: undefined reference to `extGetDevType'

RTimothyEdwards, 2022-02-08:
Considering that magic is now at version 8.3.263, let me know if any recent version is failing for you.  I compile regularly on Ubuntu, Debian, CentOS, and Fedora.

jarvis-lorry, 2022-02-08:
Hi Timothy, I've tried compiling with version 8.3.263 and I get the same problem as before, where a number of undefined references are logged during "make" and then "make install" fails.  Here are my steps:
2025  ./configure --prefix=$PWD
2026  make 
2027  make install

Here are the errors from make.log:
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:828: undefined reference to `ExtCurStyle'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:840: undefined reference to `DBIsContact'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:845: undefined reference to `DBFullResidueMask'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:848: undefined reference to `DBTypePlaneMaskTbl'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:846: undefined reference to `DBNumTypes'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:851: undefined reference to `DBNumUserLayers'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:853: undefined reference to `DBResidueMask'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:856: undefined reference to `DBTypePlaneMaskTbl'
/usr/bin/ld: /home/lekan/magic-8.3.263/extflat/EFantenna.c:854: undefined reference to `DBNumTypes'
collect2: error: ld returned 1 exit status
make[2]: *** [../rules.mak:44: extcheck] Error 1

RTimothyEdwards, 2022-02-08:
@jarvis-lorry :  Are those _all_ the error messages or just the final ones?  Can you post the entire make.log file?

jarvis-lorry, 2022-02-08:
Those were just the first few errors reported during make.  I've attached make.log
[make.log](https://github.com/RTimothyEdwards/magic/files/8025521/make.log)
Thank you.

RTimothyEdwards, 2022-02-08:
Looks like there are some places in the code which cause a failure if magic is not compiled with Tcl/Tk support.  However, magic is basically crippleware if you don't compile it with Tcl/Tk support, so I suggest you enable that by installing the prerequisite packages.

Be sure to do "make clean" before recompiling when switching from a non-Tcl-interpreter based build to a Tcl-interpreter based build.

jarvis-lorry, 2022-02-08:
Thank you for your help.  I've been able to install magic once I installed the tk development package, tk-dev.  I didn't know that “tk-dev” is separate from “tk”; I also didn't notice this warning sent out when I ran ./configure:

can't find Tk configuration script "tkConfig.sh"
Reverting to non-Tcl compilation

RTimothyEdwards, 2022-02-08:
I'm reopening this because the underlying issue still exists, which is that magic will not compile without the Tcl/Tk option.  Whether or not magic is "crippleware" without it, as I like to say, it is still supposed to maintain that backwards compatibility.  A handful of recent commits failed to honor the backwards compatibility, and I'm still trying to track down a few of them.

RTimothyEdwards, 2022-02-08:
I have restored the ability to compile the non Tcl/Tk version, for whatever use it is, although I had to drop the original build of ext2spice and ext2sim as independent executables and fold them into the program as commands, like they are in the Tcl/Tk version, because there are now too many dependencies in those two modules on code outside of the extflat module.

closed 2022-02-08 by RTimothyEdwards

davehaglan, 2022-07-17:
I had this same issue on Ubuntu 20.04 building magic from git.  Installing the packages below got me going.

apt-get install csh
apt-get install tcl-dev
apt-get install tk-dev