RTimothyEdwards/magic #134: Segfault

kop316, 2022-02-22:
Hello!

I am getting a segfault when I run this script:

```
#! /usr/bin/env bash
set -euo pipefail
#=========================================================================
# mflowgen-run
#=========================================================================
# Generator : /home/ctalbot/skywater/stanford_ee272/mflowgen/mflowgen/core/build_orchestrator.py

# Pre

rm -f .time_end
date +%Y-%m%d-%H%M-%S > .time_start
MFLOWGEN_STEP_HOME=$PWD
export design_name=GcdUnit

# Commands

cp inputs/adk/magicrc .magicrc
magic -noconsole -dnull run_antenna.tcl | tee outputs/antenna_check.log

# Post

cd $MFLOWGEN_STEP_HOME
date +%Y-%m%d-%H%M-%S > .time_end
```

run_antennd.tcl is:
```
# Copied from https://github.com/efabless/caravel/blob/master/gds/antenna_on_gds.tcl

gds read inputs/design_merged.gds

load $::env(design_name)

select top cell
extract do local
extract no capacitance
extract no coupling
extract no resisitance
extract no adjust
extract unique
extract

feedback save ./outputs/antenna.feedback.txt

antennacheck debug
antennacheck

quit
```

kop316, 2022-02-22:
Here is the gds

RTimothyEdwards, 2022-02-23:
Thanks, I should be able to debug from the posted GDS file.

RTimothyEdwards, 2022-02-23:
extflat/EFantenna.c line 252 should be `EFFlatDone(NULL)` and not `EFFlatDone()`.

I am doing some other bug fixes in magic, but I'll push that fix in a bit.  Thanks for the example!

kop316, 2022-02-23:
> extflat/EFantenna.c line 252 should be `EFFlatDone(NULL)` and not `EFFlatDone()`.
> 
> I am doing some other bug fixes in magic, but I'll push that fix in a bit. Thanks for the example!

Awesome, thank you for the fast turnaround!