RTimothyEdwards/magic #137: Commit 505155497ebbf13ada2dbe788bc5704f22ca3d7b seems to have modified spice extraction to cause LVS issues

kop316, 2022-02-24:
Commit 505155497ebbf13ada2dbe788bc5704f22ca3d7b seems to have modified spice extraction to cause LVS issues. This occurs with DEF files and GDS files.

The very same flow with commit f8390b78f88d72e3d51f8a226e15d75bccf23b33 passes LVS.

I believe the following artifacts should be able to point you in the direction of the issue (in each folder, `mflowgen-run` is the script being run).
[Artifacts.zip](https://github.com/RTimothyEdwards/magic/files/8136683/Artifacts.zip)

If you need anything else or have questions, please let me know!

kop316, 2022-02-25:
This seems to have gotten worse. I now get a segfault when trying to extract spice files from the def

RTimothyEdwards, 2022-02-25:
I prefer segfaults.  They're much easier to track down and fix.  I will look at this shortly.  Please check against the _most_ recent version of magic, if you have time.  Otherwise I will take a look later today.

kop316, 2022-02-25:
> I prefer segfaults. They're much easier to track down and fix. I will look at this shortly. Please check against the _most_ recent version of magic, if you have time. Otherwise I will take a look later today.

I can confirm the segfault happens with commit db4fa65bfc096e63954b37b188ea27b90ab31839, which AFAIK is the latest version.

RTimothyEdwards, 2022-02-25:
Okay, I bumped up the priority because this is a segfault issue.  But I could not get this to segfault.  from the results.zip file, I did:
```
magic -d OGL -rcfile /usr/share/pdk/sky130A/libs.tech/magic/sky130A.magicrc
lef read /usr/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
def read GcdUnit.def
extract do local
extract all
ext2spice lvs
ext2spice
```
What did you do to get the segfault?