RTimothyEdwards/magic #138: Latest Magic throws DRC violations

vijayank88, 2022-02-25:
Hi
Please refer below issue.
https://github.com/The-OpenROAD-Project/OpenLane/issues/954
The latest magic version throws following error, which i'm not faced with magic commit 0bb6ac1fa98b5371c73156b6e876925397fb7cbc
```
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 6
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: user_proj_example
Run Directory: /openlane/designs/FCNet_prj/runs/RUN_2022.02.22_06.52.45
----------------------------------------

Magic DRC Summary:
Source: /openlane/designs/FCNet_prj/runs/RUN_2022.02.22_06.52.45/reports/finishing/drc.rpt
Violation Message "Metal3 minimum area < 0.24um^2 (met3.6) "found 6 Times.
Total Magic DRC violations is 6
----------------------------------------
```

RTimothyEdwards, 2022-02-25:
You are reporting vs. a version of magic from 64 commits ago, version 8.3.209, using a PDK that for the past several months has contained a fixed requirement that magic must be at least version 8.3.260.  So you need to provide much more information to distinguish between what might be an error in magic, or open_pdks, or openlane, or OpenROAD.  I have not modified code around magic's handling of DRC rules for a very long time, so I am very doubtful that the issue is with magic.  What proof do you have that there are _not_ metal3 minimum area violations in the layout?

vijayank88, 2022-02-26:
Thanks @RTimothyEdwards. I will debug more on this and comeback.

vijayank88, 2022-03-31:
with OpenLane tag [2022.03.30_02.55.46](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.03.30_02.55.46), i'm not seeing those DRC violations. Hence closing this issue. Thanks @RTimothyEdwards

closed 2022-03-31 by vijayank88