RTimothyEdwards/magic #145: Absurd capacitance extraction with default Sky130 tech file.

donnie-j, 2022-03-23:
Almost surely user error on my part, but there isn't good documentation on which scale factor or setting needs to be adjusted...

Reproduce:

```
$ magic -rcfile $PDK_ROOT/sky130A/libs.tech/magic/sky130A.magicrc \
sky130A/libs.ref/sky130_fd_sc_hd/mag/sky130_fd_sc_hd__dlygate4sd3_1.mag
```

Then in magic

extract style ngspice(orig)
extract
ext2spice

The resulting file has the correct nets and transistors (equivalent to the PDK spice version of the cell), but the parasitic capacitance is huge, one is some 34nF.  In a few other attempts, trying some settings and scale factors produce even larger capacitances.  One even produced large -ve capacitance values.

Identical values result when starting from the .gds of the same cell (which is reassuring :)

This is magic version 8.3 rev 277.
J.
[sky130_fd_sc_hd__dlygate4sd3_1-1.spice.gz](https://github.com/RTimothyEdwards/magic/files/8329569/sky130_fd_sc_hd__dlygate4sd3_1-1.spice.gz)

RTimothyEdwards, 2022-03-23:
Thanks for pointing that out.  This is related to a change in the way fringe capacitance is calculated, but it looks like something is not handled correctly that didn't show up in my tests.

donnie-j, 2022-03-23:
@RTimothyEdwards many thanks, I confirm that this fix results in reasonable looking capacitance values, and simulation waveforms :)

RTimothyEdwards, 2022-04-05:
Since this fix is confirmed, I'll close this issue now.