RTimothyEdwards/magic #149: make all fails silently if database.h fails to build

proppy, 2022-03-29:
assuming `csh` is not present:
```
# ./configure
...
checking for csh... no
...
```

`make all` will fails silently and exit with `0`:
```
# make all
make all
--- errors and warnings logged in file make.log
--- making header file database/database.h
# echo $?
echo $?
0
```

while `make.log` contains the appropriate error message:
```
--- making header file database/database.h
./scripts/makedbh database/database.h.in database/database.h
/bin/bash: ./scripts/makedbh: /bin/csh: bad interpreter: No such file or directory
make[1]: *** [Makefile:48: database/database.h] Error 126
make[1]: Leaving directory '/tmp/1/magic'
```

RTimothyEdwards, 2022-03-29:
But if the lack of "csh" now prevents "configure" from completing, then it doesn't matter, does it?

RTimothyEdwards, 2022-03-29:
It's very likely that the reason that I did the grep for "Error" was because the compile was not stopping on an error, which made it hard to find the needle in the output haystack.   I'd suppose that with the change to halt the build on an error, there's no longer any particular need to filter the terminal output.

proppy, 2022-03-29:
+1, something like:
```
diff --git a/Makefile b/Makefile
index e1d61fb..dc0efc4 100644
--- a/Makefile
+++ b/Makefile
@@ -18,12 +18,10 @@ INSTALL_CAD_DIRS = windows doc ${TECH}
 all:        $(ALL_TARGET)
 
 standard:
-        @echo --- errors and warnings logged in file make.log
-        @${MAKE} mains 2>&1 | tee -a make.log | egrep -i "(.c:|Stop.|---)"
+        @${MAKE} mains
 
 tcl:
-        @echo --- errors and warnings logged in file make.log
-        @${MAKE} tcllibrary 2>&1 | tee -a make.log | egrep -i "(.c:|Stop.|---)"
+        @${MAKE} tcllibrary
 
 force: clean all
 
```
should allow the error logs and status code to propagate nicely (in combination with #148 fixes.

RTimothyEdwards, 2022-03-29:
I can see the need to remove the `egrep` statement, but is there any particular reason _not_ to tee to a log file?

proppy, 2022-03-29:
I think in that case you'd want to either:
- add `-o pipefail` to `SHELL`
- add `|| exit ${PIPESTATUS[0]}`
to make sure you propagate the `make` exit code back to the parent make invocation; otherwise it will just propagate `tee` success.

RTimothyEdwards, 2022-03-29:
Ah, right, the obscure pipefail thing.

RTimothyEdwards, 2022-03-29:
@proppy :  Given that SHELL is set to `/bin/sh`, which is not necessarily `bash`, is it better to insist that SHELL be `bash`, or just forget about logging the output?

proppy, 2022-03-29:
maybe it's best to keep the `Makefile` simple and document in the `README` how to get it to log a file if needed with `make 2>&1 | tee make.log`, wdyt?

RTimothyEdwards, 2022-03-30:
Should be all fixed now.