RTimothyEdwards/magic #162: magic 8.3.301 fails to build open_pdks w/ sky130A

proppy, 2022-05-11:
getting the following error when the conda-eda CI tries to build `open_pdks.sky130a` with magic 8.3.301:
```
2022-05-11T14:01:55.6939333Z Magic 8.3 revision 301 - Compiled on Wed May 11 12:11:10 UTC 2022.
2022-05-11T14:01:55.6939983Z Starting magic under Tcl interpreter
2022-05-11T14:01:55.6940403Z Using the terminal as the console.
2022-05-11T14:01:55.6940997Z Using NULL graphics device.
2022-05-11T14:01:55.6941507Z Processing system .magicrc file
2022-05-11T14:01:55.6942132Z Sourcing design .magicrc for technology sky130A ...
2022-05-11T14:01:55.6942563Z 2 Magic internal units = 1 Lambda
2022-05-11T14:01:55.6943208Z Input style sky130(vendor): scaleFactor=2, multiplier=2
2022-05-11T14:01:55.6943874Z The following types are not handled by extraction and will be treated as non-electrical types:
2022-05-11T14:01:55.6944527Z     ubm 
2022-05-11T14:01:55.6944928Z Scaled tech values by 2 / 1 to match internal grid scaling
2022-05-11T14:01:55.6945570Z Loading sky130A Device Generator Menu ...
2022-05-11T14:01:55.6946086Z Using technology "sky130A", version 1.0.310-0-gd7faec2
2022-05-11T14:01:55.6946708Z CIF input style is now "sky130(vendor)"
2022-05-11T14:01:55.6947242Z Warning: Calma reading is not undoable!  I hope that's OK.
2022-05-11T14:01:55.6947951Z Library written using GDS-II Release 3.0
2022-05-11T14:01:55.6948374Z Library name: sky130_ef_io__addendum
2022-05-11T14:01:55.6948984Z Reading "sky130_fd_io__simple_pad_and_busses".
2022-05-11T14:01:55.6949396Z Traceback (most recent call last):
2022-05-11T14:01:55.6950025Z   File "../common/foundry_install.py", line 1695, in <module>
2022-05-11T14:01:55.6950435Z     cwd = destlibdir,
2022-05-11T14:01:55.6951053Z   File "../common/foundry_install.py", line 277, in subprocess_run
2022-05-11T14:01:55.6951495Z     raise SystemError("".join(emsg))
2022-05-11T14:01:55.6952199Z SystemError: Command magic failed with exit code: -11
2022-05-11T14:01:55.6952671Z   magic -dnull -noconsole
2022-05-11T14:01:55.6953226Z Input script was:
2022-05-11T14:01:55.6953777Z ---------------------------------------------------------------------------
2022-05-11T14:01:55.6954416Z #!/usr/bin/env wish
2022-05-11T14:01:55.6954880Z #--------------------------------------------
2022-05-11T14:01:55.6955502Z # Script to generate .mag files from .gds    
2022-05-11T14:01:55.6955993Z #--------------------------------------------
2022-05-11T14:01:55.6956593Z crashbackups stop
2022-05-11T14:01:55.6956929Z drc off
2022-05-11T14:01:55.6957449Z locking off
2022-05-11T14:01:55.6957808Z gds datestamp 1650294714
2022-05-11T14:01:55.6958357Z gds readonly true
2022-05-11T14:01:55.6958712Z gds drccheck false
2022-05-11T14:01:55.6959254Z gds flatten true
2022-05-11T14:01:55.6959600Z gds rescale false
2022-05-11T14:01:55.6960150Z tech unlock *
2022-05-11T14:01:55.6960577Z # Set the GDS input style to sky130(vendor).  This treats labels on the
2022-05-11T14:01:55.6961271Z # TXT purpose (5) as pins, which is unfortunately done in a lot of the
2022-05-11T14:01:55.6961707Z # vendor GDS files.
2022-05-11T14:01:55.6962257Z cif istyle sky130(vendor)
2022-05-11T14:01:55.6962697Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io__analog.gds
2022-05-11T14:01:55.6963421Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io__gpiov2_pad_wrapped.gds
2022-05-11T14:01:55.6964021Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io__bare_pad.gds
2022-05-11T14:01:55.6964796Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io__connect_vcchib_vccd_and_vswitch_vddio_slice_20um.gds
2022-05-11T14:01:55.6965510Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io.gds
2022-05-11T14:01:55.6966247Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io__disconnect_vccd_slice_5um.gds
2022-05-11T14:01:55.6966821Z gds read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/gds/sky130_ef_io__disconnect_vdda_slice_5um.gds
2022-05-11T14:01:55.6967493Z puts stdout "Annotating cells from LEF"
2022-05-11T14:01:55.6968085Z lef read $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/lef/sky130_ef_io.lef
2022-05-11T14:01:55.6968757Z puts stdout "Annotating cells from CDL/SPICE"
2022-05-11T14:01:55.6969248Z catch {readspice $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_io/cdl/sky130_ef_io.cdl}
2022-05-11T14:01:55.6970060Z puts stdout "Writing all magic database files"
2022-05-11T14:01:55.6971104Z writeall force
2022-05-11T14:01:55.6971324Z puts stdout "Done."
2022-05-11T14:01:55.6971712Z quit -noprompt
2022-05-11T14:01:55.6971850Z 
2022-05-11T14:01:55.6972111Z ---------------------------------------------------------------------------
```

See the full logs here:
https://pipelines.actions.githubusercontent.com/serviceHosts/b1561d4a-c200-469d-96ed-a5c4b56e2e69/_apis/pipelines/1/runs/1129/signedlogcontent/94?urlExpires=2022-05-11T15%3A02%3A45.7572024Z&urlSigningMethod=HMACV1&urlSignature=TZVP9GtH8wKcwRjOEzmRZsrT7WIGpmtRgEMlfSYdpN8%3D

proppy, 2022-05-11:
logging this here, but I realize this might also be an open_pdks issue :P

mithro, 2022-05-11:
Exit code `-11` is a segfault, so I think it is a Magic issue?

proppy, 2022-05-11:
yep, you're right and there are no recent open_pdks changes https://github.com/RTimothyEdwards/open_pdks/commits/master that match with this version the segfault wasn't "uncovered" by some input change.

RTimothyEdwards, 2022-05-11:
Yes, this is the right place to raise the issue.  I switched to zlib functions for reading GDS (so I can get magic to read compressed GDS) and I missed converting "rewind()" to "gzrewind()".  Fixed and testing now.  Will update the git repo as soon as it has been verified.

RTimothyEdwards, 2022-05-11:
@proppy :  Issue has been fixed, tested, verified, pushed, and mirrored to github.  Thanks for bringing it to my attention.

proppy, 2022-05-12:
Looks like this fixed the issue, thanks! https://github.com/hdl/conda-eda/runs/6397325813?check_suite_focus=true

closed 2022-05-12 by proppy