RTimothyEdwards/magic #164: make has fatal error in Mac OS monterey: 'GL/gl.h' file not found

bulent2k2, 2022-05-15:
Hello,

Here is the log from ./scripts/configure_mac :

> -----------------------------------------------------------
> Configuration Summary (principle requirements):
> 
> X11:          yes
> Python3:      yes
> OpenGL:       no
> 
>   Cairo graphics are available so OpenGL is purely optional.
> Cairo:        yes
> Tcl/Tk:       yes
> -----------------------------------------------------------

I tried running magic. But, I get: 

> error in slave eval:
> dlopen(/usr/local/lib/magic/tcl/tclmagic.dylib, 0x0005): tried: '/usr/local/lib/magic/tcl/tclmagic.dylib' (no such file)
> Main console display active (Tcl8.6.12 / Tk8.6.12)
> (magic) 1 % 

Below is the 'fatal error' message from make.

Thanks for your help!
Bulent

> make[2]: Nothing to be done for `depend'.
> ../rules.mak:56: Depend: No such file or directory
> gcc -Wno-error=implicit-function-declaration   -I/usr/local/Cellar/tcl-tk/8.6.12_1/include -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.12_1/include -I/usr/local/opt/cairo/include -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -DX11 -DXLIB -DCAIRO -DNDEBUG -MM grMain.c grLock.c grDStyle.c grText.c grCMap.c grClip.c grGlyphs.c grNull.c W3Dmain.c grX11su1.c grX11su2.c grX11su3.c grX11su4.c grX11su5.c grTk1.c grTk2.c grTk3.c grTk4.c grTk5.c grTOGL1.c grTOGL2.c grTOGL3.c grTOGL4.c grTOGL5.c grTkCommon.c grOGL1.c grOGL2.c grOGL3.c grOGL4.c grOGL5.c X11Helper.c grX11thread.c grTCairo1.c grTCairo2.c grTCairo3.c grTCairo4.c grTCairo5.c | \
> 	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
> 	> Depend
> grTOGL1.c:22:10: fatal error: 'GL/gl.h' file not found
> #include <GL/gl.h>
>          ^~~~~~~~~
>

RTimothyEdwards, 2022-05-15:
You either need to install an OpenGL development package on your system or else disable OpenGL in the compile (`configure --without-opengl` should be sufficient).

RTimothyEdwards, 2022-05-15:
Not sure why this isn't handled automatically by the configure script, though.  So I might have to fix this in the configure script.

cheyao, 2023-02-07:
> Not sure why this isn't handled automatically by the configure script, though. So I might have to fix this in the configure script.

I also had this problem before, the solution for me is to use gcc as the C/C++ compiler
You need to run:
```
brew install gcc
```
and for the configure use
```
CC=gcc-12 CXX=g++-12 ./scripts/configure_mac
```
(By the way not recommending to set the env variable CC and CXX to gcc-12 with export, since that can brake some other program's builds)