RTimothyEdwards/magic #165: OASIS read/write support

antonblanchard, 2022-05-16:
GDSII files are quite large and this can cause performance problems with git and can also hit the file size limits on github. Compression (which magic recently added) definitely helps. Here is an example of the GDS produced when taping out the Microwatt core:

```
620M    user_project_wrapper.gds
123M    user_project_wrapper.gds.gz
 51M    user_project_wrapper.gds.xz
```

While not supported by either magic or klayout, xz compression shows somewhat of a lower bound on what compression could achieve.

The same file in OASIS format is significantly smaller at all levels:

```
 60M    user_project_wrapper.oas
 24M    user_project_wrapper.oas.gz
 18M    user_project_wrapper.oas.xz
```

RTimothyEdwards, 2022-05-18:
I'd be happy to integrate any kind of compression support;  however, zlib is well-developed, and with a few helpful pointers I was able to just replace various I/O routines with their zlib equivalents.  I don't know anything about how to incorporate either xz compression or OASIS directly into the code.

I like to point out that the fact that GDS is so compressible is mainly an indication that it is a truly terrible data format.  Nearly everything about it is just wrong for efficiently representing mask-level data.

mithro, 2022-05-18:
@RTimothyEdwards - IIRC `bzip` and `lzma` (xz) both have drop in compatible replacement functions for the zlib equivalent. It should be as updating the `open` / `read` / `seek` / `close` calls.

stefanottili, 2022-05-19:
Please keep in mind that every tool on the planet reads gds.gz files directly, whereas I know of no other widely used compression format for gds.
The 2005 oasis is as optimal for layout data as a format can be, so please don’t bother implementing any other compression scheme, just add oasis support. 
E.g. klayout has a gpl‘ed oasis reader/writer.

stefanottili, 2022-05-19:
From what I can tell, there are two reasons why OpenLane gds are so unusually big that are more important to address then switching to oasis, because they affect both the same way.

1) P&R (placement + routing) data is written by magic.
Magic's tiled data structure in effect flattens via and combines and fractures them with the routing path. Compared to a straight def2gds/oasis conversion with via reference and path, this blows up the gds/oasis file size.
One easy solution is to use klayout to collect P&R def and leaf cell gds + fill data.

2) OpenLane metal fill is flat.
Common practice of commercial fill tools is to use SREF or even better AREF's to represent tile data. Usually these structures also use several levels of hierarchy, drastically reducing the data and thus file size of tiling data.

So the low hanging fruit is to switch from magic to klayout to collect the P&R and IP gds and to use it to write oasis.
And then take advantage of hierarchy when creating fill data.

stefanottili, 2022-05-19:
Since we’re at the topic of using git for revision control, here’s my take on it:

Git is a tool written for source code revision control, where history is useful.
It’s taking advantage of the fact that incremental changes to source code can be efficiently stored as deltas.

None of the large, generated files in eda flows fit that description.

Gatelevel verilog, def, spef, gds/oasis and p&r databases can’t be stored as deltas and should ideally be gzipped anyways. Use the internal zlib compression for oasis.

Even more importantly, the only files worth keeping are the last version used for tapeout. During development there is a sliding window of a week or two where old files might still be in use.
But using git to keep dozens if not hundreds of intermediate versions around forever doesn’t make any sense. Both diskspace and bandwidth have real cost associated with them.

Don’t misuse a source code revision tool for binary file transfer.

Knowing which verilog source revision got synthesized matters, but it’s as easy as adding the revision and a date to the derived database and filenames.

This scheme allows to delete outdated (aka more then a week old) generated files.

maliberty, 2022-05-19:
I agree about git for flow generated gds.  For flow inputs (eg IO gds) then it makes sense but they tend to change less often.

OR metal fill doesn't use SREF/AREF but I agree it would be a good idea.  In some of the advanced nodes I see the fill becomes so complex in terms of heterogenous shapes and packing that I don't think you will get much compression from them, but in open areas they are very effective.

maliberty, 2022-05-19:
Sadly DEF has no support for fill arrays unless you also generate a side LEF to fill masters to instantiate

stefanottili, 2022-05-20:
$$$$ extraction tools use def + gds/oasis tiles for cap/res extraction, I've not encountered a flow where def FILL has been used, fill data is usually created by physical verification tools.

urish, 2024-11-05:
For reference: gdstk has an OASIS parser / writer implemented in ~1k LOC: https://github.com/heitzmann/gdstk/blob/main/src/oasis.cpp

stefanottili, 2024-11-05:
gpl'ed klayout oasis is ~3.5k loc each for reader/writer
https://github.com/KLayout/klayout/tree/master/src/plugins/streamers/oasis/db_plugin

Regarding my earlier comment with regard to fill:
def FILL makes sense if the P&R tool takes net timing criticality into account when adding fill.

maliberty, 2024-11-05:
It is a goal to consider timing in fill in OR but not a reality.