RTimothyEdwards/magic #167: Problem on extracting hierarchy cell

Park-Hyung-Joo, 2022-06-03:
I extracted dff cell with inverter, tri-state-inv and  tri-state-inv_small. Since I set port only on mosfet, I expect a netlist of dff consists of mosfet. But extracted spice file has .subckt tri-state-inverter_small which I didn't set port and dff netlist only contains tri-state-inv_small. No mosfet around tinv_small included in netlist of dff
this is netlist script
[dff_netlist.txt](https://github.com/RTimothyEdwards/magic/files/8834083/dff_netlist.txt)
this is schemtic
[image]
and this is layout
[image]

dff netlist should include mosfet in inverter and net with port of dff but it dosen't.
I used following command in magic

extract all
ext2spice lvs
ext2spice

d-m-bailey, 2022-06-03:
@Park-Hyung-Joo magic needs to have the diffusion and poly at the same level of the hierarchy to recognize (extract) devices. Could you post your mag (or gds) data too?

Park-Hyung-Joo, 2022-06-04:
mag data of mosfets
[nmos_nf2_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836962/nmos_nf2_mag.txt)
[pmos_nf2_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836963/pmos_nf2_mag.txt)
[nmos_2stack_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836966/nmos_2stack_mag.txt)
[pmos_2stack_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836967/pmos_2stack_mag.txt)

various inverters ( I expected them didn't appear as .subckt because they don't have ports only label)
[inv_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836972/inv_mag.txt)
[tinv_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836973/tinv_mag.txt)
[tinv_small_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836974/tinv_small_mag.txt)

and dff data ( top cell )
[dff_mag.txt](https://github.com/RTimothyEdwards/magic/files/8836976/dff_mag.txt)

all these files are mag format and real file name of them are
nmos_nf2_mag -> nmos13_fast_center_nf2.mag
pmos_nf2_mag -> pmos13_fast_center_nf2.mag
nmos_2stack_mag -> nmos13_fast_center_2stack.mag
pmos_2stack_mag -> pmos13_fast_center_2stack.mag
inv_mag -> inv_2x.mag
tinv_mag -> tinv_2x.mag
tinv_small_mag -> tinv_small_1x.mag
dff_mag -> dff_2x.mag

Park-Hyung-Joo, 2022-06-04:
I tried to extract latch circuit consist of inverter and nand and now ext2spice extract nothing. Only way to extract is flatten whole circuit.

latch cell is made by inv and nand subcell
[image]

ext file says it use inv and nand (inv has port and nand has no port)
[image]

and inv and nand include pmos_center and nmos_center_cell
[image]

as you can see spice from nand cell has pmos_cell and nmos_cell ( spice form inv also has)
[image]

But latch cell extract only nmos_cell and pmos_cell as subckt and not even include them in top cell netlist
[image]

RTimothyEdwards, 2022-06-16:
@Park-Hyung-Joo :  Sorry for the delay in response.  Getting a flattened extraction when you have standard cells inside your standard cell requires a specific set of options, as follows:

```
ext2spice lvs
ext2spice hierarchy off
ext2spice subcircuit off
ext2spice subcircuit top on
```

This is probably more complicated than it needs to be, but anyway, it will do what you want.

RTimothyEdwards, 2022-06-16:
Magic does not actually care whether a circuit has ports or not.  If a circuit makes a connection into a subcircuit, that connection is treated as a port.  Only by setting the ext2spice options can you control whether the extraction is or isn't hierarchical.  There is a method to individually treat subcircuits as black boxes by substituting an "abstract view" of the subcell;  but here, you just wanted to flatten everything, so using the ext2spice option settings is the best way.