RTimothyEdwards/magic #168: Incorrect pwell connections for standard cells in dnwell

d-m-bailey, 2022-06-04:
Magic 8.3 revision 308
Tech file version 1.0.291-20-g05af1d0

MPW-5 slot-009 armleo gpio mpw5 test chip

[image]

There are standard cells placed in dnwell region. The pwell connections (VNB) for some standard cells are not being connected for some standard cells. Notice the `<instance>/VNB` nets.
```
Xsky130_fd_sc_hd__nand2_4_3 vss vdd armleo_gpio_lv2hv_3/A sky130_fd_sc_hd__inv_4_1/Y
+ out_l sky130_fd_sc_hd__nand2_4_3/VNB vdd sky130_fd_sc_hd__nand2_4
Xsky130_fd_sc_hd__nand2_4_2 vss vdd sky130_fd_sc_hd__inv_4_1/A oe_l med_enable sky130_fd_sc_hd__nand2_4_2/VNB
+ vdd sky130_fd_sc_hd__nand2_4
Xsky130_fd_sc_hvl__inv_4_1 vssio vddio sky130_fd_sc_hvl__inv_4_1/Y armleo_gpio_lv2hv_1/Y
+ vssio vddio sky130_fd_sc_hvl__inv_4
Xsky130_fd_sc_hd__nand2_4_4 vss vdd armleo_gpio_lv2hv_2/A oe_l out_l vss vdd sky130_fd_sc_hd__nand2_4
Xsky130_fd_sc_hd__inv_4_0 vdd vss sky130_fd_sc_hd__inv_4_0/Y sky130_fd_sc_hd__inv_4_0/A
+ sky130_fd_sc_hd__inv_4_0/VNB vdd sky130_fd_sc_hd__inv_4
Xsky130_fd_sc_hd__inv_4_1 vdd vss sky130_fd_sc_hd__inv_4_1/Y sky130_fd_sc_hd__inv_4_1/A
+ sky130_fd_sc_hd__inv_4_1/VNB vdd sky130_fd_sc_hd__inv_4
Xsky130_fd_sc_hd__inv_4_2 vdd vss armleo_gpio_lv2hv_4/A_N armleo_gpio_lv2hv_4/A sky130_fd_sc_hd__inv_4_2/VNB
+ vdd sky130_fd_sc_hd__inv_4
Xarmleo_gpio_ggnmos_0 pad vddio vssio armleo_gpio_ggnmos
Xsky130_fd_sc_hd__inv_4_4 vdd vss armleo_gpio_lv2hv_3/A_N armleo_gpio_lv2hv_3/A sky130_fd_sc_hd__inv_4_4/VNB
+ vdd sky130_fd_sc_hd__inv_4
Xsky130_fd_sc_hd__inv_4_5 vdd vss armleo_gpio_lv2hv_1/A_N armleo_gpio_lv2hv_1/A sky130_fd_sc_hd__inv_4_5/VNB
+ vdd sky130_fd_sc_hd__inv_4
Xsky130_fd_sc_hd__inv_4_6 vdd vss armleo_gpio_lv2hv_0/A_N armleo_gpio_lv2hv_0/A sky130_fd_sc_hd__inv_4_6/VNB
+ vdd sky130_fd_sc_hd__inv_4
Xarmleo_gpio_pfet_driver_x9_0 sky130_fd_sc_hvl__inv_16_5/Y vddio pad armleo_gpio_pfet_driver_x9
Xsky130_fd_sc_hd__inv_4_8 vdd vss armleo_gpio_lv2hv_5/A_N armleo_gpio_lv2hv_5/A sky130_fd_sc_hd__inv_4_8/VNB
```
To duplicate, extract the attached tarball, and
```
./run_ext armleo_gpio armleo_gpio.gds.gz
```
[armleo_gpio.tar.gz](https://github.com/RTimothyEdwards/magic/files/8836579/armleo_gpio.tar.gz)

RTimothyEdwards, 2022-06-04:
I encountered a similar problem in a DAC design that I did that I surrounded with deep nwell.  My hypothesis is that if a deep nwell is large enough, the cookie-cutter extraction method in magic isn't dealing with internal  areas of the deep nwell, where it does not see the connecting nwell.  But I haven't looked into it yet.

RTimothyEdwards, 2022-06-16:
@d-m-bailey :  I expected to find some commonality in the instances that are not connected, but they seem to be selected at random.  The VNB ports of some cells just don't appear anywhere in the .ext file, where every one of them should be in a "merge" line.

RTimothyEdwards, 2022-06-16:
Once again, sorry for the late reply.  I've been rather busy for the last two weeks.  This one came in while I was on travel, so it got put on the back burner.  I'm looking into it carefully now.

RTimothyEdwards, 2022-06-16:
There is a point of commonality.  If the "cookie cutter" square being extracted doesn't contain one of the taps, then the VNB port of any subcells falling within the same square will be lost.  That should be enough information to go on to fix this error.

RTimothyEdwards, 2022-06-17:
One hack solution to this (while I'm busy trying to actually fix it properly) is to change the extraction step size in the magic tech file;  the existing step size is 7um (`step 7` in section `extract` in `sky130A.tech`).  If you increase this to, say, 15, then there should always be a tap in the extraction square, and the issue should disappear.