RTimothyEdwards/magic #171: SKY130A tech file error

fener1efes2, 2022-07-16:
I cannot open SKY130A.tech file and magic gives the following error

[image]

d-m-bailey, 2022-07-16:
The magic tech file is defined in the magicrc file. If there is a `.magicrc` file in your working directory, it will use that. You can also specify the file on the command line `magic -rcfile $PDK_ROOT/sky130A/libs.tech/magic/sky130A.magicrc`

fener1efes2, 2022-07-16:
What does `$PDK_ROOT` and `$CAD_ROOT` mean?
I installed magic on an external SSD.
Directory of magic is located at `/media/user/SSD/magic`
Where is my working directory located at or `$CAD_ROOT/magic/sys` ?

RTimothyEdwards, 2022-07-16:
@fener1efes2 :  The solution from Mitch Bailey assumes that you have installed the SkyWater PDK using the open_pdks installer;  the installation location passed to the open_pdks configure script is $PDK_ROOT.  The command line shown in the solution assumes you have set $PDK_ROOT as an environment variable.  If you used the default configuration for open_pdks, then $PDK_ROOT would be /usr/local/share/pdk/.  This has nothing to do with where magic itself is installed.

closed 2022-08-30 by fener1efes2