RTimothyEdwards/magic #173: proto.magicrc.in:1:3: error: invalid preprocessing directive

yurivict, 2022-07-29:
Build fails with errors:
```
In file included from <built-in>:355:
<command line>:61:9: warning: 'CAD_DIR' macro redefined [-Wmacro-redefined]
#define CAD_DIR /usr/local/lib
        ^
<command line>:6:9: note: previous definition is here
#define CAD_DIR "/usr/local/lib"
        ^
proto.magicrc.in:1:3: error: invalid preprocessing directive
#       $(CAD_ROOT)/magic/sys/.magicrc 
        ^
proto.magicrc.in:2:4: error: invalid preprocessing directive
#       System wide start up file for magic, defines default macros.
        ^
proto.magicrc.in:5:3: error: invalid preprocessing directive
#       Source file proto.magicrc.in
        ^
proto.magicrc.in:6:3: error: invalid preprocessing directive
#       Process this file with the cpp macro processor
        ^
proto.magicrc.in:13:4: error: invalid preprocessing directive
#  Default .magicrc macro file (new macros)
   ^
```

Re-starting the build after this succeeds.

This suggests that there is possibly a dependency problem in a parallel build.

yurivict, 2022-07-29:
Single-CPU build also has the same issue.

RTimothyEdwards, 2022-07-30:
This comes from the magic/Makefile recipe
```
proto.magicrc: proto.magicrc.in
        ${MCPP} ${GR_DFLAGS} ${DFLAGS} proto.magicrc.in > proto.magicrc
```

where `${MCPP}` is defined in the `defs.mak` file as
```
MCPP                  = /bin/python3 ${MAGICDIR}/scripts/preproc.py -ccomm
```

This value is set from the `configure` script if you have `python3` on the system;  then everything works great.  If you don't have python3, then it defaults to the `cpp` processor, and the behavior of `cpp` is very system-dependent (which is why I replaced it with the python script).

yurivict, 2022-07-30:
Maybe configure should fail when python3 isn't available?

RTimothyEdwards, 2022-07-30:
@yurivict :  Well, `cpp` does work on a number of systems.  I just don't know which systems have an incompatible version of `cpp`.