RTimothyEdwards/magic #174: About RC PEX file extraction using magic command

Elon-J, 2022-08-05:
When I use the command to extract a PEX file with RC, it doesn't work; it just has C in my PEX file. How can I solve this problem? It said that 'Cannot open file (UNNAMED).sim' attach the result and my extract file. The log file is in output.log, the GDS file is PMU_top_level, and the magic command is in run_lvspex.sh. Thank you! @msaligane

[image]

[image]

RTimothyEdwards, 2022-08-05:
The above output depends on a script `run_lvspex.sh` which is not in the zip file.

RTimothyEdwards, 2022-08-05:
There are a number of issues in the GDS itself, such as layers that don't exist in the sky130 process.

I am getting an error from the `extresist` command where it seems to be confused by `$` in names.  I will look into that.  That does seem to prevent the generation of the `.res.ext` file which would explain why the output only has parasitic capacitors in it.

Elon-J, 2022-08-06:
Hi Tim, sorry, I use the command below to run the extraction for R-C (the run_lvspex.sh script), Do I need to flatten the GDS first to use the resistor extract command? Thank you very much!

magic -rcfile ./sky130A/sky130A.magicrc -noconsole -dnull << EOF
gds read $1
load $2
extract do local
extract all
ext2sim labels on
ext2sim
extresist tolerance 10
extresist
ext2spice lvs
ext2spice cthresh 0
ext2spice extresist on
ext2spice -o ./EXTRACT_PEX/$2_pex.spice
quit -noprompt
EOF

RTimothyEdwards, 2022-08-07:
At the moment, yes, you need to flatten the circuit:
```
magic -rcfile ./sky130A/sky130A.magicrc -noconsole -dnull << EOF
gds read $1
load $2
select top cell
flatten $2_flat
load $2_flat
select top cell
extract do local
extract all
ext2sim labels on
ext2sim
extresist tolerance 10
extresist
ext2spice lvs
ext2spice cthresh 0
ext2spice extresist on
ext2spice -o ./EXTRACT_PEX/$2_pex.spice
quit -noprompt
EOF
```

It is possible to rename the flattened circuit back to the original name, but the above script is simpler.  Just bear in mind that the name of the top level subcircuit in the output file has the "_flat" extension.

Elon-J, 2022-08-07:
thank you very much, Tim! It's really helpful, let me try it again.

RTimothyEdwards, 2022-08-08:
If the circuit is flattened, then your PEX file should not have a subcircuit called DAC_COMP.

Flattening with no options will cause the subcircuit pins to become labels with the name of the subcircuit instance prepended;  e.g., pin `VOUT` becomes label `DCDC_DAC$2_0/VOUT`.  Since that is an ordinary label, then the top level pin `D[1]` will take precedence.

Elon-J, 2022-08-09:
I see, Let me try just to use these top-level pins to test, thank you very much, Tim!

Elon-J, 2022-08-15:
Hi, Tim, when I try to extract the whole RC PEX file for our circuit, I got an error. If the magic collapse when extracting this circuit? Or I don't use the right command? Attach the screenshot, gds file, magic command, and magic running log file. The gds file is in top_level.zip, the magic command is in run_lvspex.zip and the magic running log file is in output.log. Thank you very much!
[image]

[top_level.zip](https://github.com/RTimothyEdwards/magic/files/9334633/top_level.zip)
[run_lvspex.zip](https://github.com/RTimothyEdwards/magic/files/9334636/run_lvspex.zip)
[output.log](https://github.com/RTimothyEdwards/magic/files/9334638/output.log)

Elon-J, 2022-08-16:
@RTimothyEdwards, Hi, Tim, I have a problem when I want to extract an RC PEX file for my top.gds file. The error is just like this:
[image]
[image]

How can I solve it? Is this caused by too many labels when flattening the GDS file? Thank you very much! Attach the whole running package directory for your reference:
[extract_PEX.tar.gz](https://github.com/RTimothyEdwards/magic/files/9354382/extract_PEX.tar.gz)

RTimothyEdwards, 2022-08-22:
Sorry for the delay;  I was rather busy last week.  I'm looking at it now and I can confirm the crash condition.

RTimothyEdwards, 2024-08-09:
Use:
```
extract all
ext2sim labels on
ext2sim
extresist all
ext2spice lvs
ext2spice cthresh 0.1
ext2spice extresist on
ext2spice
```

Note that usually one does `extresist tolerance 10; extresist` instead of `extresist all` but this layout is very small and nothing in it is considered by extresist to have a resistance large enough to be worth breaking into multiple nets.

daipvd, 2024-08-12:
Thank you, @RTimothyEdwards 
I have followed your guide and I got resistance extracted now.

dai