RTimothyEdwards/magic #177: About magic install problem

Elon-J, 2022-08-10:
When I try to install the magic on a new machine, I meet the 'vector fonts' problem, and when I do 'make', there are some errors. How can I solve it? Thank you! Attach the './configure' problem and 'make' problem.
[image]
[image]

RTimothyEdwards, 2022-08-10:
From the "system requirements" section of the "Install" web page for magic:

```
sudo apt-get install mesa-common-dev libglu1-mesa-dev
```

On OS versions like Fedora the library names will be different;  I don't recall off the top of my head what they are.

Elon-J, 2022-08-10:
Hi, Tim, I have installed it, but still have the above problems.
[image]

RTimothyEdwards, 2022-08-10:
From the screenshot, you installed `libGL` but not `libglu1`, which is a separate library.

RTimothyEdwards, 2022-08-10:
Since you are on a system that uses yum, the library you need is `mesa-libGLU-devel`.

Elon-J, 2022-08-15:
Thank you, Tim! It works well!

closed 2022-08-15 by Elon-J