RTimothyEdwards/magic #178: installation problem in macOS(monterey)

vinayrayapati, 2022-08-20:
I followed the instructions in the macOS install file. I made the changes to the makefile before I run make . when I finally run magic I am getting the below errors:

```
application-specific initialization failed: no display name and no $DISPLAY environment variable
Error in startup script: no display name and no $DISPLAY environment variable
    while executing
"load /usr/local/opt2/tcl-tk/lib/libtk8.6.dylib Tk"
    ("package ifneeded Tk 8.6.10" script)
    invoked from within
"package require Tk $tcl_version"
    invoked from within
"if {$tcl_version < 8.0} {
    return -code error "tkcon requires at least Tcl/Tk8"
} else {
    # Prevent breaking on version 8.5.2
    # package requ..."
    (file "/usr/local/lib/magic/tcl/tkcon.tcl" line 44)
```

what should I do?

vinayrayapati, 2022-08-20:
And when I follow the instructions with homebrew I am getting the following error in magic console:
```
error in slave eval:
dlopen(/usr/local/lib/magic/tcl/tclmagic.dylib, 0x0005): symbol not found in flat namespace (_GrTCairoPlotSVG)
```

Can you help me resolve both the above issues.

d-m-bailey, 2022-08-20:
@vinayrayapati Do you have Xquartz installed and running?

vinayrayapati, 2022-08-20:
yes I do

d-m-bailey, 2022-08-20:
This is how I configured magic
```
./configure --with-tcl=/usr/local/opt2/tcl-tk/lib --with-tk=/usr/local/opt2/tcl-tk/lib --x-includes=/opt/X11/include --x-libraries=/opt/X11/lib CFLAGS=-Wno-error=implicit-function-declaration
```

After installing tcl/tk following the instructions in `INSTALL_MacOS.md`.

vinayrayapati, 2022-08-20:
@d-m-bailey  I did the same!

d-m-bailey, 2022-08-20:
In your `magic/scripts/config.log` what are the values of `HAVE_LIBCAIRO` and `MAGIC_WRAPPER`?

vinayrayapati, 2022-08-20:
@d-m-bailey  
1 and 1

d-m-bailey, 2022-08-20:
Same here.
This is from my `config.log`
```
extra_defs=' -DCAD_DIR=\"${LIBDIR}\" -DBIN_DIR=\"${BINDIR}\" -DTCL_DIR=\"${TCLDIR}\"'
extra_libs=' ${MAGICDIR}/ext2sim/libext2sim.o ${MAGICDIR}/ext2spice/libext2spice.o ${MAGICDIR}/calma/libcalma.o ${MAGICDIR}/cif/libcif.o ${MAGICDIR}/plot/libplot.o ${MAGICDIR}/lef/liblef.o ${MAGICDIR}/extflat/libextflat.o ${MAGICDIR}/garouter/libgarouter.o        ${MAGICDIR}/mzrouter/libmzrouter.o ${MAGICDIR}/router/librouter.o       ${MAGICDIR}/irouter/libirouter.o ${MAGICDIR}/grouter/libgrouter.o       ${MAGICDIR}/gcr/libgcr.o ${MAGICDIR}/tcltk/libtcltk.o'
gr_cflags=''
gr_dflags=' -DX11 -DXLIB -DOGL -DCAIRO'
gr_hprog=''
gr_hsrcs=''
gr_libs=' -lX11 -lGL -lGLU    -lm -lcairo -lfontconfig -lfreetype ${X11_LDFLAGS}'
gr_srcs=' ${TK_SRCS} ${TOGL_SRCS} ${TCAIRO_SRCS} ${TKCOMMON_SRCS}'
```
Same for you?

vinayrayapati, 2022-08-20:
difference in the gr_cflags.
```
extra_defs=' -DCAD_DIR=\"${LIBDIR}\" -DBIN_DIR=\"${BINDIR}\" -DTCL_DIR=\"${TCLDIR}\"'
extra_libs=' ${MAGICDIR}/ext2sim/libext2sim.o ${MAGICDIR}/ext2spice/libext2spice.o ${MAGICDIR}/calma/libcalma.o ${MAGICDIR}/cif/libcif.o ${MAGICDIR}/plot/libplot.o ${MAGICDIR}/lef/liblef.o ${MAGICDIR}/extflat/libextflat.o ${MAGICDIR}/garouter/libgarouter.o 	${MAGICDIR}/mzrouter/libmzrouter.o ${MAGICDIR}/router/librouter.o 	${MAGICDIR}/irouter/libirouter.o ${MAGICDIR}/grouter/libgrouter.o 	${MAGICDIR}/gcr/libgcr.o ${MAGICDIR}/tcltk/libtcltk.o'
gr_cflags=' -I/opt/homebrew/opt/cairo/include'
gr_dflags=' -DX11 -DXLIB -DOGL -DCAIRO'
gr_hprog=''
gr_hsrcs=''
gr_libs=' -lX11 -lGL -lGLU    -lm -lcairo -lfontconfig -lfreetype ${X11_LDFLAGS}'
gr_srcs=' ${TK_SRCS} ${TOGL_SRCS} ${TCAIRO_SRCS} ${TKCOMMON_SRCS}'
```

d-m-bailey, 2022-08-20:
I don't think I installed homebrew.

vinayrayapati, 2022-08-20:
but that is because of the homebrew installation method , I don't think that affects anything, does it?

vinayrayapati, 2022-08-20:
if I do the without brew installation I am having the same logs as you!
```
extra_defs=' -DCAD_DIR=\"${LIBDIR}\" -DBIN_DIR=\"${BINDIR}\" -DTCL_DIR=\"${TCLDIR}\"'
extra_libs=' ${MAGICDIR}/ext2sim/libext2sim.o ${MAGICDIR}/ext2spice/libext2spice.o ${MAGICDIR}/calma/libcalma.o ${MAGICDIR}/cif/libcif.o ${MAGICDIR}/plot/libplot.o ${MAGICDIR}/lef/liblef.o ${MAGICDIR}/extflat/libextflat.o ${MAGICDIR}/garouter/libgarouter.o 	${MAGICDIR}/mzrouter/libmzrouter.o ${MAGICDIR}/router/librouter.o 	${MAGICDIR}/irouter/libirouter.o ${MAGICDIR}/grouter/libgrouter.o 	${MAGICDIR}/gcr/libgcr.o ${MAGICDIR}/tcltk/libtcltk.o'
gr_cflags=''
gr_dflags=' -DX11 -DXLIB -DOGL -DCAIRO'
gr_hprog=''
gr_hsrcs=''
gr_libs=' -lX11 -lGL -lGLU    -lm -lcairo -lfontconfig -lfreetype ${X11_LDFLAGS}'
gr_srcs=' ${TK_SRCS} ${TOGL_SRCS} ${TCAIRO_SRCS} ${TKCOMMON_SRCS}'
```

d-m-bailey, 2022-08-20:
Does magic work? Or do you get the same error message?

vinayrayapati, 2022-08-20:
@d-m-bailey I am getting the same error message

d-m-bailey, 2022-08-20:
What does `echo $DISPLAY` give you in your konsole?

vinayrayapati, 2022-08-20:
@d-m-bailey nothing!

d-m-bailey, 2022-08-20:
Is the Xquartz icon in your task bar?

vinayrayapati, 2022-08-20:
yes

d-m-bailey, 2022-08-20:
You run `xman` from the menu or `xeyes` from the konsole?

vinayrayapati, 2022-08-20:
if I run `xeyes` from terminal it shows:
```
Error: Can't open display: 
```

vinayrayapati, 2022-08-20:
same for the xman too
```
Error: Can't open display: 
```

d-m-bailey, 2022-08-20:
Did you reboot after installing Xquartz?

vinayrayapati, 2022-08-20:
No

d-m-bailey, 2022-08-20:
Try rebooting.

vinayrayapati, 2022-08-20:
@d-m-bailey  the magic is running but I am getting an error in the tkcon console:
```
error in slave eval:
dlopen(/usr/local/lib/magic/tcl/tclmagic.dylib, 0x0005): symbol not found in flat namespace (_GrTCairoPlotSVG)
```

vinayrayapati, 2022-08-20:
The second issue that I posted at the start!

d-m-bailey, 2022-08-20:
Only I can think to suggest is to delete the magic directory,  reclone, and try to rebuild it. Sorry.