RTimothyEdwards/magic #186: extract unique notopports not working as expected

d-m-bailey, 2022-09-28:
Version 8.3.322

Expecting `extract unique`  and `extract unique notopports` to give the same results for all subcircuits expect the top level ports.

However, subcircuits with unconnected wires that have the same text are extracted as connected.

In the top view, you can see `inp` and `inn`.
[image]

However, the child cell, `bgfc_diffpair_p`, has incorrectly labeled both wires `inn`.
[image]

Using `extract unique notopports`, I expect that the child nets to be extracted as `inn` and `inn_uq0`, but they are merged in the extracted netlist. Subcells are extracting differently with `notopports` when the expectation is that they will be extracted the same as `extract unique`.

[image]

To duplicate, download the following tarball and execute the following:
```
tar xzf test_ext.tar.gz
cd test_ext
./run_ext
vimdiff *.ext/*gds.spice
```
[test_ext.tar.gz](https://github.com/RTimothyEdwards/magic/files/9661590/test_ext.tar.gz)

RTimothyEdwards, 2022-09-28:
@d-m-bailey :  Really dumb error.  A one-word modification to magic version 8.3.328 fixes it.  Thanks for providing the example!

RTimothyEdwards, 2022-10-07:
Should be fixed now. . . can this issue be closed?

d-m-bailey, 2023-03-24:
I believe this works now.

closed 2023-03-24 by d-m-bailey